RespiraWorks/Ventilator #879: Dear Team, we are using MPX5700AP/GP/GP1 CASE 867B-04 gauge pressure sensor in the oxygen line we think to use PC2, PC3 pins cycle controller we have a doubt that in the existing circuit we are using a scaling circuit. please suggest whether the same scaling, the filter circuit can be used for the gas sensor or not please suggest if any modifications required Thank you.

saipraveenpanchakarla, 2020-08-26:
Please take the time to describe the bug in detail, so other contributors can pick it up and implement or declare done/obsolete in case it was fixed out of band.

**Current behaviour**
* Describe what is wrong with the system as of writing: what is the actual state of things.
* Instructions to reproduce the bug.
* Screenshots, screencasts, or video to help explain your problem.

**Expected behaviour**
* Describe what is the expected state of things.
* Identify documents that need to be updated, especially in case of hardware component choices or changes assembly.

**How do you know it is wrong**

Link any documents, Slack threads, quote personal communication, reference your personal domain expertise etc. explaining why you think the current state of things is bad and the expected state of things is good.

**Code pointers**

Give some thoughts on roughly where in code do you think the necessary changes might be located.

**Don't forget**
- [ ] Put the issue in a milestone
- [ ] Assign it to one of the projects
- [ ] Add appropriate labels
- [ ] Mention dependent/blocked issues if any

saipraveenpanchakarla, 2020-08-26:
Dear Team, we are using MPX5700AP/GP/GP1
CASE 867B-04 gauge pressure sensor in the oxygen line we think to use PC2, PC3 pins cycle controller we have a doubt that in the existing circuit we are using a scaling circuit. please suggest whether the same scaling, the filter circuit can be used for the gas sensor or not please suggest if any modifications required Thank you.

martukas, 2020-08-26:
Duplicated following the proper template in #880

closed 2020-08-26 by martukas