RespiraWorks/Ventilator #882: UPS & BATTERY SELECTION

Rafee-UM, 2020-08-27:
**What has to be done**
We need to finalize the UPS/Battery capacity required for Current ventilator. And need to resource it for the medical graded one which is easily available for supply chain.

**How do you know it has to be done**

**Don't forget**
- [ ] Put the issue in a milestone
- [ ] Assign it to one of the projects
- [ ] Add appropriate labels
- [ ] Mention dependent/blocked issues if any

inceptionev, 2020-08-28:
I recommend picking a unit that has the specific UPS capability and at least 200W continuous of power delivery capability to provide margin, more if you go to the higher power blower.  You need something that can do a seamless switchover when mains power fails, so in this case, either of the CyberPower units will work.

inceptionev, 2020-08-28:
You can also make an internally-integrated solution, and the beginnings of such a design are discussed here: #642

Rafee-UM, 2020-09-04:
We have reviewed one of the Ventilator in AMTZ. They are simply using Li-ION battery for 1 Hour backup. We feel, we can also make a trail with Li-Ion battery which was made in India. We have ordered it to one of our vendor and they made it in 10 cm length X  7cm height X 7cm width. And it's capacity is 10AH @12V. Stress test for 14V & 12A is going on currently.

martukas, 2021-01-20:
@tijanim can we "leave space" for this in v0.4 build?

inceptionev, 2021-01-20:
@martukas @tijanim here is a 1hr battery that I think will fit nicely below the pneumatics tray.  Leaving an envelope for this battery should be good:
https://a.aliexpress.com/_mqwvGUH

An alternate, if you have clearance issues:
https://a.aliexpress.com/_mO8e6sl

Stretch goal:  if we have room to support a lead-acid option, this is what it might look like:
https://www.batteryspace.com/sealedleadacidbattery12v10ah120whs.aspx or 2x of these: https://www.batteryspace.com/Sealed-Lead-Acid-Battery-6V-12AH.aspx

inceptionev, 2021-01-29:
@martukas @afarrington17 update on the battery.  For Li-Ion, we want to switch over to this battery (about the same dimensions as the first battery):
https://www.aliexpress.com/item/1005001883671852.html
Alternate for clearance issues (about the same size as the second battery): 
https://www.aliexpress.com/item/4001036484239.html

For sealed-lead acid, it stays the same.

inceptionev, 2021-04-05:
Drafted a battery mounting solution.  Uses PAPR-style elastic cage to hold battery into a fitted base.  Base has holes for mcmaster 94180A351 melt-in brass threaded inserts for M4 screws.  Both models and assembly in this zip: [LiPo_4s3p_BatteryAndHolder_39x57x133mm.zip](https://github.com/RespiraWorks/Ventilator/files/6255973/LiPo_4s3p_BatteryAndHolder_39x57x133mm.zip)

[image]

inceptionev, 2021-04-06:
Adding models for the other battery configuration with holder base: [image]

[LiPo_4s3p_Battery_57x67x75mm.zip](https://github.com/RespiraWorks/Ventilator/files/6261308/LiPo_4s3p_Battery_57x67x75mm.zip)

martukas, 2021-04-06:
The latter one seems like the best fit:
[image]
[image]
[image]

martukas, 2021-04-06:
This is good enough to prove that enclosure can accommodate it without modifications in metal. Kicking this to next milestone.

martukas, 2022-06-21:
This part needs to be redesigned with proper parametrization and to meet our CAD standards - feature, variable naming, etc..