RespiraWorks/Ventilator #888: Update Enclosure Drawings for Sheet Metal Thickness

DonnaCG, 2020-08-30:
Currently, the manufacturing drawings for enclosure are missing thickness dimension for all components.
The dimension should be 1.587mm or 16GA stainless.

The following drawings need to be updated:
04-0003-A
04-0004-A 
04-0007-A

- [ ] update git repo to point the latest manufacturing drawings

- [ ] update google drive/[mechanical](https://drive.google.com/drive/u/3/folders/1ofOnLvz3SM6yLp8f28dm5VryIbWRP3OO)

this update can be completed after beta build design review, to include other CAD updates.

tijanim, 2021-01-19:
@KishanKoradiya might wanna take a look at this, this could get rolled into #943 as well

martukas, 2021-02-06:
I am not sure what the basis for this suggestion is. The current enclosure -- and I measured -- is 1.52mm.

tijanim, 2021-02-06:
I think she wanted to make sure were following standard sheet metal thickness for 16GA. In any case, We can update but I doubt this would be an issue

martukas, 2021-02-15:
Sheet metal thickness has been identified on the enclosure page as 1.52mm. This has been merged in #1018 
Let's consider this documented and in case there is a well-argued reason to use different thickness sheet metal, that design change will need to come in as a PR that encompasses CAD and documentation updates.

closed 2021-02-15 by martukas