RespiraWorks/Ventilator #890: Design an external circuit for indication.

saipraveenpanchakarla, 2020-08-31:
**What has to be done**
Its shows the functionality of the ventilator i) power indication ii) functionality indication iii) bettary low/high indication.
.
**How do you know it has to be done**
We get feedback from some medical experts and also we refer iso standards.

**Hardware pointers**
We put 5 pin connector to the next revision pcb and make an external board with leds for indication.

dcstraney, 2022-06-20:
Actually ignore that, going to make it a lot bigger just for the text: the text should be at least 10 mm high (4x the size) to be readable when your face isn't right up against the front panel.

dcstraney, 2022-06-20:
Ok this should be better, as long as there's room for it:
[image]

martukas, 2022-06-20:
Ooops, I just saw this having asked for similar in the PR.

So, the display panel mounts with M2.5 -- maybe we could use the same for this one? They are less obtrusive and would look more consistent.

Height - the (protruding part of the) touch screen is 100mm tall. We could try to match that for consistency?
Width - the total space available is 340, screen takes up 165. So We are left with potentially up to 175, other than mounting.

However, maybe we do want to leave some open acrylic to peek through, so we shouldn't hog as much as we can. Smaller might be fine? This is a kind of debug device, and the clinician will likely focus their attention on the screen most of the time. How about using some compact icons to indicate the functions?

dcstraney, 2022-06-20:
Ok, was only using M4 because it was the "preferred default" for fasteners so M2.5 sounds even better here for such a small board, will change that!

Smaller outline definitely makes sense for looking @ LEDs etc. on the main board - wasn't sure of the exact use case for these LEDs, but if it's to be used mostly for debug then don't have to worry about spelling things out completely.  I can't think of any intuitive icons that would fit "UI computer" and "cycle controller", but can start with the first layout above (the smaller one) and shrink things a bit more with just "UI" and "CC" labels (plus the smaller holes).

martukas, 2022-06-20:
Yes, you are correct that M4 is our "most preferred" fastener, but just a bit below that is also M2.5 A few spots where those have been unavoidable, so we are in the M4+M2.5 regime (mostly).

"UI" and "CC" seems reasonable. If you google images of "power icon" you'll get the typical circle broken with a vertical line. I suppose cycle could be something like a circle with arrows. Not sure what UI might be...

dcstraney, 2022-06-20:
Yeah I had the same thoughts with the power symbol - see what you think about this, nice and compact now (bonus standards-compliance: the vertical LED arrangement matches a traffic light):
[image]

martukas, 2022-06-20:
Cool! This looks pretty good. Are the LEDs intended to protrude or stay behind the glass? If there isn't anything to physically interact with, I wonder if  it could even be mounted somehow without through-holes at all... @gokcemay ideas?

dcstraney, 2022-06-20:
Stay behind the glass: the idea (to keep things simple) was just to use the same SMT LEDs that we use as status indicators on the main board already, and take advantage of the clear acrylic.  We could always do light pipes or something in the future if there's a need for an opaque and "polished-looking" front panel.

Best I can think of for the icons is a chip for the cycle controller (to symbolize the internal operation) and a screen for the UI (10-second sketch below).  Still don't like that very much for the cycle controller though.
[image]

Anyways I'll go ahead and update the branch.  No idea what's up with the code coverage test.

closed 2022-06-23 by martukas