RespiraWorks/Ventilator #892: Adding a Barb on the filter housing on the tube side.

RakurthiLokesh, 2020-09-01:
Please take the time to describe the issue in detail, so that other contributors can pick it up and implement or declare it done/obsolete in case if was completed out of band.

**What has to be done**
A barb has to be added to the filter housing to hold the tube firmly as the tube is getting slipped from the filter housing.

**How do you know it has to be done**
While assembling the enclosure we faced an issue of the tube gets slipped off from the filter housing so form the next revision we are planning to add a barb on the housing so that it will help in holding the tube firmly.
[image]

**Don't forget**
- [ ] Put the issue in a milestone
- [ ] Assign it to one of the projects
- [ ] Add appropriate labels
- [ ] Mention dependent/blocked issues if any

martukas, 2020-09-04:
Question about the shape of the barb. Could it not instead be like this type of barb that we have on most of tube fittings?
[image]

I am concerned that width of the "notch" must be compatible with whatever clip we use to attach the tubing, so this may be making too much of at assumption at this stage in prototyping?

martukas, 2020-11-04:
This issue is superseded by #917

closed 2020-11-04 by martukas