RespiraWorks/Ventilator #925: Ensure support for alternative blowers

martukas, 2020-11-20:
**What has to be done**
PCB and electrical system in general should support alternative blowers that we may test with in the next iteration prototype.
We have the following on hand:
* [24V TKFAN BA7050H24B](http://www.tkfan.com/Hot_Sell/ba7050-22.html), no driver board included
* [WM7060-24V](https://usa.banggood.com/WM7060-DC-12V24V-High-Pressure-Blower-12Kpa-Brushless-DC-Blower-Small-Centrifugal-Fan-For-sleep-Ventilator-Oxygen-Treatment-p-1436015.html?ID=6232653&cur_warehouse=CN)
*[WM9290-24V](https://www.aliexpress.com/i/10000048332654.html)
* The 2 above have come with driver board included which is also labeled "WM9290-24V" like the latter of the two blowers

It would also be interesting if we could support the "from scratch" blower designed by these guys:
https://github.com/makers-for-life/makair
I have not been able to easily find the electrical side of their design

**How do you know it has to be done**
@inceptionev has noted that our current blower (WS7040) was not delivering enough pressure in full system with the 4 in line air filters installed. There were also indications that we may not be able to deliver the 60cm under all lung conditions described by standards.

**Code pointers**

**TODO: not sure which part of the PCB design this will affect, someone from the appropriate team should edit this to provide necessary info**

**Don't forget**
- [ ] Put the issue in a milestone
- [ ] Assign it to one of the projects
- [ ] Add appropriate labels
- [ ] Mention dependent/blocked issues if any

inceptionev, 2020-11-20:
We should run tests to confirm this, but both the 12V and 24V drivers should have the same control interface, just a different input voltage, and either of them should be able to run the TKFAN.

martukas, 2020-11-22:
The 2 TKFANs I have look like this:
[image]

martukas, 2021-01-18:
Outcomes might influence #713

martukas, 2021-01-18:
This is related to mechanical design adaptations for supporting alternative blowers #940

dcstraney, 2021-01-19:
Blowers mentioned above (as well as original blower) all supported through their external driver boards as of #947

closed 2021-01-19 by dcstraney