RespiraWorks/Ventilator #932: Detailed circuit analysis

dcstraney, 2020-12-13:
Need to analyze the design in more detail for mostly thermal stresses, and some add'l details.  Will put and distribute together some sort of standard spreadsheet for doing these so that we can easily recalculate when we know internal temps, etc.

5V buck supply:
- [ ] Control loop stability analysis (current values are just copied from datasheet example, but w/different Cout)
- [ ] Inductor behavior in fault (high ambient temp, output @ current limit)
- [ ] TPS54561 power dissipation in normal (?) and fault (high ambient temp, output @ current limit) conditions; have to make assumptions about internal temp., can figure out necessary heatsinking this way

Solenoid driver:
- [ ] TPS1H100 power dissipation in normal and fault conditions; have to make assumptions about internal temp., can figure out necessary heatsinking this way
- [ ] MOSFET power dissipation in normal and fault conditions; "

Stepper motor driver:
- [ ] powerSTEP01 power dissipation in normal and fault conditions; "

Input power handling/battery interface:
- [ ] Power dissipation of various components (?) in normal and fault conditions

Analog sensor inputs:
- [ ] RTD reading uncertainty @ STM, based on amplifier error, resistor tolerance + tempco, STM's ADC error
- [ ] Pressure reading uncertainties "
- [ ] O2 reading uncertainties "

inceptionev, 2021-02-27:
started power/thermal calcs here: https://docs.google.com/spreadsheets/d/1Otu4uRq1WR7CHx7AOSxF2eoAqoWKeeAT3UDNmPhiSdw/edit?usp=sharing

inceptionev, 2021-03-04:
VSYS Buck under 20-ohm resistive load.  We were worried about subharmonic oscillation from a confusing equation in the LM5141 datasheet that did not seem to close with any reasonable inductor values, so we are testing this on an EVM altered with our design compensation components and selected inductor.

Here we see no evidence of subharmonic oscillation at the battery nominal voltage of 14.4V, when the converter is already switching way above a 50% duty cycle.  We start to see a little bit of it as we get down to 13.0V, but the output regulation stays stable and constant, so this seems to be benign.  No audible buzzing from the converter either.  At 13V we would consider a Li-Ion battery basically dead, less than 10% charge remaining.

We also see that the regulator continues to work as the voltage drops, all the way down to about 12.5V, when the switching frequency drops and the converter is essentially just conducting.  The output voltage starts to droop to in the input voltage at this point, but the converter doesn't show any problematic behavior even as the voltage drops well below the battery dead voltage.

[image]

Next we will try this using the 72W BA7050H12B blower as the load.

inceptionev, 2021-03-04:
Here are traces with the 72W blower running at full blast, 6A of load.  The power supply wiring at my bench is too small for this, and you'll see a lot of input voltage drooping, to the point of the controller dropping in and out of regulation, but it seems to handle this without any adverse behavior.

There isn't a lot of headroom between a 14.4V battery and the 12V system voltage.  This seems to work fine though.  I wonder if we should consider a buck-boost (either this or the next revision)

[image]