RespiraWorks/Ventilator #934: Pneumatics Tray Acrylic Modeling and Re-Design

tijanim, 2020-12-14:
**What has to be done**
The Pneumatics tray needs to be updated to reflect new manifolds and unnecessary holes. See the following image for a revised layout: 

[image]

The following shall be created:
- Inventor Solid Model
- Engineering Drawing (both .idw and exported PDF)
- Dxf file (For laser cutting)

**Don't forget**
- [ ] Put the issue in a milestone
- [ ] Assign it to one of the projects
- [ ] Add appropriate labels
- [ ] Mention dependent/blocked issues if any

martukas, 2020-12-14:
The two big holes on the right should not be there. The new more central ones are the new positions and the old ones are not needed.

KishanKoradiya, 2020-12-15:
Status update: Inventor solid model (Screeenshot below) and DXF files ready. Engineering Drawing remains, tolerances TBD and title block template to be updated. 
[image]

KishanKoradiya, 2020-12-16:
Here are the updated files. Some points to be noted:

- Corner hole punches not at center of fillets
- Circle dimensions almost but not exactly same and not standard values (e.g. diameter of 30.48)
- I have no idea of the tolerances. ±0.25 mm seems reasonable
- A lot of stuff in the title block is empty, since I wasn't sure what to put in there (e.g. QA, Manufacturing, Dwg. No.)
- I will create a branch pertaining to this issue add these models there since this page does not support all types of files. PDF has been attached though.
[[Shelf Acrylic] V3_710-00002_ebk v1_DWG.pdf](https://github.com/RespiraWorks/Ventilator/files/5699424/Shelf.Acrylic.V3_710-00002_ebk.v1_DWG.pdf)

PR #979 (2021-02-03): v0.3 Pneumatics Tray Redesign
Redesign of the pneumatics tray to v.3 components. Closes #934
closed 2021-02-03 by martukas