RespiraWorks/Ventilator #937: Revise Enclosure Page to Include Acrylic Sheets and Assembly Components

tijanim, 2020-12-15:
**How do you know it has to be done**
The Enclosure page is straight up sad right now. A lot of information on how to put together this build is missing and makes understanding components not directly contained in other assemblies quite difficult.

**What has to be done**
The enclosure page needs to be expanded to include all components present in the current build outside of what is covered by other assemblies. It should include:

- Revised metal enclosure instructions and drawings
- A general assembly and/or sectional drawing detailing assembly of the entire unit
- Acrylic sheet CAD and drawings (Face Plate, Filter Plate, Pneumatics Tray)
- Pneumatics Assembly overview drawing
- LOM for any hardware required
- Updated Assembly CAD (if possible) 

**References**
- Beta CAD (Fusion)
- Beta BoM: https://docs.google.com/spreadsheets/d/1i01Zh0hXRvrTphOLACNOmgj-qOsmSzOZ/edit#gid=2003192236
- Slack # eng-mechanical channel

**Don't forget**
- [ ] Put the issue in a milestone
- [ ] Assign it to one of the projects
- [ ] Add appropriate labels
- [ ] Mention dependent/blocked issues if any

martukas, 2021-01-18:
The scope of this is to encompass the 0.3 enclosed version but not improvements beyond that.

martukas, 2021-02-15:
Repurposing this ticket as leaping board for some general refactoring of v0.3 CAD and BOM.

Currently we need to tie up these loose ends:
* explain choice of hardware, preference for diameters and drivers/heads
* move techniques / methods guides to their own directory
  * 3d printing
  * heat insert installation
  * acrylic manufacture and modification techniques
* add logo decal to enclosure
* add main power plug to closure panel
* explain optional wiring of blower to interim board (#926)
* finish migrating all screws to button head / hex drive
* re-import some third-party parts now that we have tacit understanding for CAD use
* remove unused third-party parts
* make most of electrical assembly part of display panel assembly
* pneumatic assembly should now be "internals assembly", with whatever directory restructuring it might precipitate
* pneumatic assembly needs more refactoring into sub-assemblies, in CAD as well as in readmes
* add newest photo of v0.3 assembly
* mention deburring/filing/sanding of sharp corners in enclosure
* list bottoming instead of tapered tap for venturi nipples
* add some supplier and part number info to parts metadata (depending on feedback from efforts in #1011 )

Most likely will not be tackled here:
* sensor tubing CAD
* general wiring CAD
* tubing clips in CAD
* general update of all drawings
* replacement of Tygon tubing & fittings in oxygen line (#996)
* including cooling fan (#1034)
*

martukas, 2021-02-17:
Most of the CAD modeling content has been addressed through #1035 

Remaining issues are captured in:
* sensor tubing #1039 
* main power input #1040 
* cooling fan #1034 
* custom elbows #1019 
* final drawings and exports #1041 

This ticket is superseded by the above more granular and better defined tasks.

closed 2021-02-17 by martukas