RespiraWorks/Ventilator #940: Update blower inlet design

inceptionev, 2020-12-24:
**What has to be done**
Update blower inlet design to make it more secure, easier to assembly, and more easily adapted to different blower models.

**How do you know it has to be done**
Current design works, but requires precise simultaneous alignment of multiple pieces that are held together by the same set of fasteners.  In particular, the blower and two gaskets are not captive to each other during assembly, thus they have to be temporarily held by clamps during assembly.  The parts can become misaligned during assembly or subsequent operation, and it might not be immediately visible.

[image]

**Code pointers**
This is the inlet adapter on the blower.

**Proposed course of action**  
Inlet adapter that is registered to the blower itself during and after assembly by a captive o-ring.

It is registered to the blower inlet and sits floating in the hole in the acrylic pneumatic panel, the inlet opening of the blower facing downwards in the orientation of the screenshots below.

A different adapter could be made to adapt the design for different blower models.

Below is a conceptual prototype that requires refinement:
[image]
[image]
[image]
[image]
Note: in the last picture above the blower and adapter are oriented upside-down with respect to how they would be sitting in the pneumatic plate.

In particular, this design was tested but the assembly force felt too high, and the blower edge sits a bit above the acrylic when fully assembled so the inlet pocket can probably be a little bit deeper.  In general though, the concept seems to work.

**Don't forget**
- [ ] Put the issue in a milestone
- [ ] Assign it to one of the projects
- [ ] Add appropriate labels
- [ ] Mention dependent/blocked issues if any

tijanim, 2020-12-28:
Thinking I'll finish #924 first, get that PR pushed, and then dive back to tackle this. Makes more sense to have everything defined in inventor first

inceptionev, 2020-12-28:
Sounds good, this one is not a priority, just wanted to get it down so it’s not forgotten.

Edwin

martukas, 2021-01-18:
This is related to similar ticket for PCB support for various blowers #925 
And may also influence outcome of final blower selection #713

martukas, 2022-10-22:
Anyone wishing to continue work on this should use the material in #1307 as the starting point.
Some design rationale and documentation about intended use and integration are provided in their students' term paper, available on our Google Drive [here](https://docs.google.com/document/d/16LkomxeedI8qgLuz92zj0P6J-rPslExN/edit?usp=sharing&ouid=105641223261656214741&rtpof=true&sd=true).

* The designs provided by SUU students should be printed and tested.
* Designs should be rebuilt from the ground up in Inventor, properly parametrized with constraints.

Some improvements already known to be needed even prior to physical testing:
* Base should integrate a tubing adapter, making it unnecessary to have additional gaskets to mate with acrylic
* The "side-panel" on the base is probably for mounting driver board that is only needed by some blower models. It should be possible to disable this feature parametrically.
* TK blower needs stand-in model
* The base could probably be made smaller in footprint
* The assembly could probably be secured with screws rather than a strap.

martukas, 2023-04-26:
Links to O-ring and libricant parts, and possible other nuances are in notes, comments and review discussion in the CAD-only update in #1307

martukas, 2023-04-26:
Links to O-ring and lubricant, and possible other nuances are in notes, comments and review discussion in the CAD-only update in #1307

martukas, 2023-04-26:
What remains for this to close is:
* drawings
* renderings
* bills of materials
* assembly instructions with pictures
* other documentation, such as design rationale and other sections required typical of mechanical assembly pages