RespiraWorks/Ventilator #941: Migrate Pinch Valve components to Inventor

KishanKoradiya, 2020-12-28:
Please take the time to describe the problem in detail, so that other contributors can pick it up and implement or declare it done/obsolete in case if was completed out of band.

**When to create Feature request or Task?**

Feature request is like a User Story. Task represents work that needs to be done. (Sub-)task would be underneath a Feature request (Story) which in our case (R&D) might just stand in the way.

Task can include non-code and most likely will i.e. study/investigate X , do a PoC of Y (pseudo-code?) , get more info about something (+writing a documentation is a task)

**What has to be done**
Describe what has to be done, ideally in declarative form - what does success look like.

**How do you know it has to be done**
Link any documents, Slack threads, quote personal communication, reference your personal domain expertise etc. explaining why you think this should be done.

**Code pointers**
Give some thoughts on roughly where in code do you think the necessary changes might be located.

**Don't forget**
- [ ] Put the issue in a milestone
- [ ] Assign it to one of the projects
- [ ] Add appropriate labels
- [ ] Mention dependent/blocked issues if any
- [ ]

KishanKoradiya, 2020-12-28:
Please note:

The [README](https://github.com/RespiraWorks/Ventilator/tree/issue_941_pinchvalve_migration/manufacturing/pinch_valve#custom-parts) has links to external webpage, as quoted below, that required account creation to view. 
   

> Original designs found [here](https://cad.onshape.com/documents/3fe0c1f79c482144c267173d/w/2ad1c08071a25185f9c78c68/e/03a49465e4e026f9f102d0af).

Suggest removing it if we no longer need it.

closed 2021-01-26 by martukas

martukas, 2021-02-05:
Superseded by #990

PR #990 (2021-01-26): Pinch valve migration to Inventor, update to v2.0, adaptation for vent build 0.3
* reimplemented CAD of all pinch valve components in Inventor
* all sketches parameterized and most constraints labeled and optimized for future modifications
* incorporates gusset supports for shelf
* rotor space in the base is tighter, for improved durability
* rotor is taller to better interface with 30mm axle bolt
* mounting screw spacing changed for improved assembly
* stop shelf distance from shaft is dictated by tube diameter and compression slack parameters
* highly parametrized for improved adaptability i [... cut]