RespiraWorks/Ventilator #952: Select oxygen regulator

martukas, 2021-01-18:
**What has to be done**
A pressure regulator needs to be selected for oxygen inlet.
Preliminarily, this unit has been ordered for 0.3 enclosed prototypes:
https://www.smcpneumatics.com/IR2010-N02B.html

Some testing should be done to confirm that it works as desired in the correct range.
Question has been raised whether one with integrated gauge might be preferred?

**How do you know it has to be done**
Need more detail...

**Code pointers**
Decision should be reflected by update to this page:
https://github.com/RespiraWorks/Ventilator/tree/master/manufacturing/oxygen

**Don't forget**
- [ ] Put the issue in a milestone
- [ ] Assign it to one of the projects
- [ ] Add appropriate labels
- [ ] Mention dependent/blocked issues if any

martukas, 2021-01-18:
Decision should likely be part of design report #787

martukas, 2021-01-18:
Selection of different unit might affect #937 and/or #938

inceptionev, 2021-01-27:
Test data from the IR2010-N02B regulator in the CA enclosed vent build has been posted here: #991 

Enclosed build sees some drop in performance vs the old pizzabuild data, which is to be expected due to all the extra resistance in the system from the added filters and check valve.  Performance is also basically on par with the old regulator placed in to the new Enclosed build.  We are still meeting the requirement of 60lpm delivery (1000cc/s) but only marginally.  I'm willing to say that this means that we can move forward with a purchase of this regulator, but in the style that is OXYGEN SAFE and includes a gauge: **IR2010-N02BG-X1**.

However, if we are willing to wait to test it out, based on the datasheet, the smaller cousin of this reg might also work, the **IR1010-N01BG-X1**
[image]

inceptionev, 2021-02-04:
[image]
**Latest Data from regulator testing**
- Tested several configurations on the enclosed build, using Test #1 which is a high flow, high volume test.
- Includes testing of the lower intermediate pressure / larger PSOL orifice configuration proposed in #993.
- We are getting basically equivalent performance between the old reg and the IR2010.
- In addition, we also get equivalent performance with the IR1010 at 45psi.
- However, the proposed lower pressure configuration from #993 did not seem to achieve the same performance.  (this will be discussed further on #993)

General observations:
- The IR2010 and IR1010 have similar performance at 45psi and both meet the requirement of 60lpm peak flow.
- A regulator with flow characteristics similar to either of these should work fine for development going forward (note that this is not a final qualification)
- **However** there is a significant problem with the IR-series regulators.  In normal operation, even when with no gas demand, there is a bleed that vents out of the regulator and is part of the precision control system of the regulator.  This bleed will accumulate 100% oxygen inside the housing of the ventilator and is probably not appropriate for use here.
- A regulator with similar characteristics but without this bleed is desired.
- Also of note, the IR2010 and IR1010 regulators are not oxygen safe.  
- SMC makes a line of cleaned regulators, without the bleed, that use fluorinated lubricants.  While they do not certify them as oxygen safe, they are probably closer to what we want.
- A possible candidate is the SRH3110-N02.  It has a performance somewhere between the IR2000 and IR1000 series.

martukas, 2021-03-25:
This has been suggested by @esc-works:
https://www.smcpneumatics.com/SRH3000-N02.html

With the caveat that there is an additional PSOL? @inceptionev can you add details about that?

inceptionev, 2021-04-07:
Here is the bracket for this regulator: https://www.smcpneumatics.com/B21-1-T1.html

martukas, 2022-07-19:
Use an adapter from #996 between regulator and  panel-mount oxygen inlet, and from regulator to PSOL.

inceptionev, 2022-07-19:
Note: in March 31 2021, we purchased 8x of these for oxygen-safe prototyping: SRH3000-N02

martukas, 2023-04-23:
Mount bracket CAD no longer available. Item appears to be discontinued.