RespiraWorks/Ventilator #954: Implement temperature sensing for motors

martukas, 2021-01-18:
**What has to be done**
There should be a way to sense stepper motor temperature and alarm when they overheat. This entails some layers of sensors, electronics and software.
This probably needs to be broken into separate issues and/or linked to existing aspects of the solution that are already being implemented.

**How do you know it has to be done**
Stepper motors tend to heat up. This may be a fire hazard.

**Code pointers**
NEED MORE INFO

**Don't forget**
- [ ] Put the issue in a milestone
- [ ] Assign it to one of the projects
- [ ] Add appropriate labels
- [ ] Mention dependent/blocked issues if any

dcstraney, 2021-01-21:
Electronics is implemented: extra inputs for 10K NTC thermistors added by Edwin to PCB Rev. A as of #947.

inceptionev, 2021-03-04:
this has been implemented in PR# 947

closed 2021-03-04 by inceptionev