RespiraWorks/Ventilator #960: Selection of mechanical power switch

martukas, 2021-01-19:
**What has to be done**
We need a mechanical power switch for powering up/off the ventilator. A mechanical switch should be chosen that:
* satisfied the electrical parameters of the system
* meets ergonomic requirements, i.e. will not accidentally be switched off easily
* meets splash/spill safety requirements

**How do you know it has to be done**
India build provided feedback that such a switch is needed.

**Code pointers**
TODO

**Don't forget**
- [ ] Put the issue in a milestone
- [ ] Assign it to one of the projects
- [ ] Add appropriate labels
- [ ] Mention dependent/blocked issues if any

inceptionev, 2021-01-20:
Let's use this switch for modeling: https://www.digikey.com/en/products/detail/e-switch/R1966CBLKBLKGF/1805114

martukas, 2021-01-21:
Does it need to be recessed in some way to prevent accidental flipping?

inceptionev, 2021-01-21:
That would be ideal, but I think we can punt this to next rev if needed.

dcstraney, 2022-06-24:
Took a look for switches, with these criteria:
- 20VDC+ rating, preferably 24VDC or more (circuit designed for 20VDC max.)
- 10ADC+ rating, more is better (circuit designed for 8.3A max. under normal conditions, but various fault conditions could raise this)
- Some level of water resistance/sealing
- Rocker switches only, as there are plenty of E-Switch "vandal-resistant" push-on/push-off latching switches, but the lack of direct visual feedback on the switch's state seems like a large potential problem in the right (/wrong) situations
- SPST preferred and with "0 / 1" markings preferred, as it's harder to screw up the connections during assembly this way
- Flange-mount or screw-body only, leaving out "snap-in" switches as these depend on a very specific panel thickness and are generally less robust in their mounting ...however I had to leave this out as nothing came up

Results were...
- NKK [JWMW series](https://www.digikey.com/en/products/detail/nkk-switches/JWMW11RA1A/671485), has an optional barrier accessory which is meant for insulation/dielectric strength but also makes the switch harder to bump.  Acceptable panel thickness is 1.0 - 4.0 mm (or up to 3.2 mm with barrier).  Barrier is available either assembled with switch, or separately as the [AT217](https://octopart.com/search?q=at217&currency=USD&specs=0&manufacturer_id=269) for more flexible sourcing.
- NKK [WR series](https://www.digikey.com/en/products/detail/nkk-switches/WR11AF/1249126)
- E-Switch [R7 series](https://www.e-switch.com/product-catalog/sealed/product-lines/r7-series-illuminated-sealed-rocker-switch#.YrT6t73MLbg) if the "no snap-in" rule is relaxed: R711...D*** or R712...D*** or R722...D***.  Currently-available part numbers which fit are: [R711FA1-SXXXNBDNNX](https://octopart.com/r711fa1-sxxxnbdnnx-e-switch-33582909?r=sp) non-illuminated, [R711FA2-SXXX2BDRNW](https://octopart.com/r711fa2-sxxx2bdrnw-e-switch-34872076?r=sp) 12VDC red illuminated, [R711FE1-SXXX2BDGNW](https://octopart.com/r711fe1-sxxx2bdgnw-e-switch-34872080?r=sp) 12VDC green illuminated, [R712FA1-SXXXNBDNNX](https://octopart.com/r712fa1-sxxxnbdnnx-e-switch-33582912?r=sp) non-illuminated.  Acceptable panel thickness is 1.5 - 4.5 mm.
- Bulgin [C1500 series](https://www.digikey.com/en/products/detail/bulgin/C1500WABB/5058059), although lack of markings isn't preferred.

There may be other switch manufacturers worth looking at, like TE Connectivity/Alcoswitch (although their website's search is absolutely unusable last time I checked) or Nidec Copal.

martukas, 2022-06-24:
Thanks for the in-depth analysis. The one with the barrier does make sense.
In terms of mounting - the acrylic is 6mm thick. Otherwise, we could mount on the aluminum, which is about 1.3mm, but then it would break our decoupling of fully functional internals assembly vs. enclosure as shell only.

dcstraney, 2022-06-24:
Ok, that's what I was afraid of with the snap-in mounting.  Sounds like for now, the solution might be to...
1. Mill the acrylic to a 3 mm depth in the area of the switch, or
2. Make a small sub-panel to mount to a hole in the acrylic?
Both are kind of obnoxious, but don't see any better alternatives.

inceptionev, 2022-06-24:
We can also consider making the back panel 3mm.  It doesn't need to hold as much weight as the pneumatic panel.

martukas, 2022-06-26:
This looks to be an acceptable solution for now.

martukas, 2022-07-17:
Looks like the

martukas, 2022-07-17:
Looks like the non-barrier type cannot be upgraded with the barrier. We have to order the specific barrier type:
https://www.digikey.com/en/products/detail/nkk-switches/JWM11BA1A-A/1058358

martukas, 2022-07-17:
And connectors for the back should be the narrower type, probably this?:
https://www.digikey.com/short/nfr9h7rj

don-superlative, 2022-07-18:
Yeah that's a strange-looking insulator but as long as it's soft enough for the crimper to squash, that looks good!

I assume there's not an easy way to add the "anti-accidental-turn-off" barrier as a feature of the existing front panel?  How much of a pain would it be to add an additional 2D-cut piece, a small "picture frame" with a screw hole or two for mounting, and 6 mm+ thick, to sit around the switch?  Would be nice to be less tied to this much-less-available one switch model when there's a bunch of other good options, when the barrier can be ignored.

martukas, 2022-07-18:
Actually, it might work out just right. We need a thinner piece of acrylic to snap this in anyway, so we can install it recessed behind the main 6mm panel, thus the same result.

don-superlative, 2022-07-18:
Ok perfect, that would be ideal then.  So might as well keep the "R" switches, not buy the "B" switches, and just return the barriers?

inceptionev, 2022-07-18:
> And connectors for the back should be the narrower type, probably this?: https://www.digikey.com/short/nfr9h7rj

yup this one will work (i.e. molex 0190020035)