RespiraWorks/Ventilator #962: Enclosure with two (2) cooling fans

martukas, 2021-01-19:
**What has to be done**
Enclosure should accommodate 2 equipment cooling fans.
* fans should not interfere with other internal pneumatic components
* fans should be effective at cooling the most heat-prone or heat-sensitive components. Some thought my have to be put into this as stepper motors are below the main equipment tray, while computing equipment is above.

**How do you know it has to be done**
India build indicated excessive heat and need for a second cooling fan.
TODO: link to notes

**Code pointers**
TODO

**Don't forget**
- [ ] Put the issue in a milestone
- [ ] Assign it to one of the projects
- [ ] Add appropriate labels
- [ ] Mention dependent/blocked issues if any

inceptionev, 2021-01-20:
If we have space constraints, consider using a more powerful fan in the same size envelope, adding holes that allow airflow to cross the pneumatic tray, and providing an outlet for the airflow that opens from the space below the tray.  This will encourage airflow across the stepper motors, which create a lot of heat.  Also pay mind to the steady state temperature of the Raspberry Pi processor, which will throttle back on performance if it overheats.  Here is one that is the same size with 2x the flow and 5x the static pressure: https://www.digikey.com/en/products/detail/nmb-technologies-corporation/08025DA-12P-AA-00/9841655

RakurthiLokesh, 2021-01-22:
We India team thought of going for two fans initially but later we decided to go for a high powered fan with vent holes to allow the air to flow on the opposite side.

martukas, 2021-04-06:
Newly designed enclosure side-panels have cut-outs and mounting holes for fans on each side. Two fans can be installed or, alternatively one of those cut-outs can be used to mounting an air filter, or can be blocked off.

The fans are placed at the back so that they maximize air-flow around the most heat sensitive PCB components.

These changes have been checked-in in the branch addressing #957 

[image]
[image]

closed 2021-04-06 by martukas