RespiraWorks/Ventilator #964: Handles on sides of enclosure

martukas, 2021-01-19:
**What has to be done**
There should be two handles on the sides of the enclosure for carrying the ventilator. They may be of whatever material as long as they do not protrude but are recessed for safety reasons.

**How do you know it has to be done**
The top handle made of bent metal likely has to be eliminated as a safety hazard, as noted in #963

**Code pointers**
TODO

**Don't forget**
- [ ] Put the issue in a milestone
- [ ] Assign it to one of the projects
- [ ] Add appropriate labels
- [ ] Mention dependent/blocked issues if any

martukas, 2021-01-21:
Here are some options from McMaster.

Recessed handles:
https://www.mcmaster.com/recessed-pull-handles/handle-style~recessed/mount-type~screw/

Folding:
https://www.mcmaster.com/recessed-pull-handles/mount-type~screw/handle-style~folding/

KishanKoradiya, 2021-03-16:
The links above have handle models that were unacceptable for various reasons - the recessed models had material that would show on the exterior of the box and was considered ugly and the folding handles were also deemed aesthetically displeasing.

These qualities were laid out as desirable:
- not ugly
- easily installable
- recessed
- protect the hand from sheet metal
- to not get stuff inside

Solution proposed to satisfy these:
A recessed [snap on](https://www.mcmaster.com/handles/mount-type~snap-in/) handle with minimal material on the outside with no screws needed for securement and is easily installable is a great choice imo:
https://www.mcmaster.com/1098A92/   (matches our sheet metal thickness - 1.52mm is approx. 0.06'')
https://www.aliexpress.com/item/4001260324420.html
https://southco.com/en_us_int/handles/flushlow-profile-handles-pulls/p2-flush-pull-lockable-and-nonlockable-versions ([full catalog](https://files.southco.com/static/Literature/p2.en.pdf))

martukas, 2021-04-06:
Going with this for now:
https://www.mcmaster.com/1098A27/

This is made for panel thickness of 0.03"
If we are using 18 gauge stainless for enclosure panels, those are 0.05", so this is not quite right. Perhaps it can be gently forced to conform?

[image]

sock8m, 2021-04-13:
https://www.mcmaster.com/1098A91/

thickness is for 0.04" to 0.055"

sock8m, 2021-04-13:
[image]

The 1098A91 handle has a different fit then the 1098A27

martukas, 2021-08-17:
Complete in #1103

PR #1103 (2021-06-09): updates: handles on the side of the enclosure
# Description

Changed the handles on side.ipt and side_left.ipt to the new size, changed handle cutouts to fit new handles.

Fixes #964 

# General checklist:

- [x] I have performed a self-review, including - looked through the `Files changed` tab, browsed repository in my branch
- [ ] I have made corresponding changes to the documentation - to reflect changes in code, electrical or mechanical design
- [ ] All new documentation or graphics follow the [documentation style guide](https://github.com/RespiraWorks/Ventilator/wiki/Documentation-style-g [... cut]
closed 2021-08-17 by martukas