RespiraWorks/Ventilator #965: Placement of power supply in enclosure

martukas, 2021-01-19:
**What has to be done**
Must decide where the power supply (adapter) and power jack are to be placed in the enclosure.
* must not interfere with pneumatic assembly
* must include necessary mounting for adapter (which adapter?)
* must have cut-out in the enclosure and secure mounting for power cable (or power jack?)

**How do you know it has to be done**
The v0.3 builds had no provision for any of this, and there was not even a way to run a power cable out of the enclosure without cutting additional holes in metal.

**Code pointers**
TODO

**Don't forget**
- [ ] Put the issue in a milestone
- [ ] Assign it to one of the projects
- [ ] Add appropriate labels
- [ ] Mention dependent/blocked issues if any

martukas, 2021-01-19:
Blocked by #966 (selection of power supply)

martukas, 2021-01-21:
We want to keep the ventilator agnostic of the AC power grid, so it will internally be 12V. Thus, no power adapter is integrated into the unit. A selection of the external adapter is tracked in #966 and the enclosure should simply have a plug such as this:
https://www.digikey.com/en/products/detail/kycon-inc/KPJX-PM-4S/9990081

tijanim, 2021-01-22:
Mirror what was done on front. Acrylic back holding PCB/Battery can also mount switch/power connector. A cutout in the back plate allows access.

martukas, 2021-04-06:
So, power supply itself is definitely external.
Having a connector on the enclosure is addressed in #1040

closed 2021-04-06 by martukas