RespiraWorks/Ventilator #966: Selection of power supply / voltage adapter

martukas, 2021-01-19:
**What has to be done**
Must decide on power supply / voltage adapter for ventilator:
* must provide enough power for PCB, valves, RPi and touchscreen
* must fit in enclosure
* must be reliable (possibly medical grade?)

**How do you know it has to be done**
TODO

**Code pointers**
TODO

**Don't forget**
- [ ] Put the issue in a milestone
- [ ] Assign it to one of the projects
- [ ] Add appropriate labels
- [ ] Mention dependent/blocked issues if any

martukas, 2021-01-20:
decision on power supply might actually be helpful in ordering 20 units for the next iteration builds?

inceptionev, 2021-01-20:
@martukas I looked at other options and it looks like a lot of medical power supplies use this connector in higher currents.  I think we should probably consider switching over to this connector at least until we know our power envelope better.

As an alternative, the MeanWell medical GSM160A12-R7B is a similar part, provides 11.5A, and is $64 in small quantities.

inceptionev, 2021-01-20:
For the input connector we can use this: https://www.digikey.com/en/products/detail/kycon-inc/KPJX-PM-4S/9990081

martukas, 2021-03-11:
@inceptionev has identified one that might fit behind the acrylic:
https://www.digikey.com/en/products/detail/cui-devices/MD-40SP/166042

dcstraney, 2021-07-11:
For future reference, that last one (the CUI MD-40SP) is unfortunately too small (mini-DIN rather than "power DIN"/Kycon's not-actually-DIN size).  It looks like the KPJX-PM-4S or KPJX-PM-4S-S (same with metal flange) are the only panel-mount options.  There is a vertical through-hole one that protrudes more, but would have to mount to a small board that mounts to the housing at a few places, and seems like a broken solder joint waiting to happen after however many hundreds of insertions and removals: https://www.digikey.com/en/products/detail/cui-devices/PD-40VS/2712204

dcstraney, 2021-07-11:
For the 14V-20V (no battery charging) or 17-20V (with battery charging) we'll need, AC adapter options available from Digikey are:
18V, $68: https://www.digikey.com/en/products/detail/globtek-inc/TR9CH9440YL4IMR6B/9837832 (don't know Glob-Tek or how trustworthy they are)
15V, $68: https://www.digikey.com/en/products/detail/xp-power/VES150PS15/7897385

inceptionev, 2021-07-11:
The DC input is current-limited, so we want to go to 19V or 20V to get the most power.  

We haven't tested any power supply yet for v0.3, but I recommend going with this one, which is in stock here:
https://www.bravoelectro.com/gsm160a20-r7b.html  (Mouser looks to have a couple in stock too)

The Glob-tek one could also work, but note that it has a different power connector pinout from the Mean-Well.

The only connector that we've tested to work so far is: https://www.digikey.com/en/products/detail/kycon-inc/KPJX-PM-4S/9990081

martukas, 2022-07-28:
Need to go with this for 24v as required by new PCB