RespiraWorks/Ventilator #993: Large swing of regulation pressure on oxygen

inceptionev, 2021-01-27:
**Current behaviour**
While collecting the data in #991, I noticed something possibly of importance. To get the 60lpm through the PSOL, we are running the regulation pressure at 45psi, just below the source pressure of 50psi.  With that little difference, there is no way any regulator is going to keep up.  There is a substantial bounce in pressure (~20psi) with each breath.  So basically, this reg is doing the work of protecting our system from unexpected high pressure, which is half the job, but not doing the other half, which is giving us consistent control performance.  The videos are too large to post here, but see Slack messages:
- 50psi Source Pressure https://respiraworks.slack.com/files/U0100QAMZL5/F01KMRS3Z1C/file_from_ios.mov
- 100psi Source Pressure https://respiraworks.slack.com/files/U0100QAMZL5/F01LJELQXUY/file_from_ios.mov

What can we do about this? Do we need to do anything about this?  Hopefully we can get a pneumatics person to chime in here. 
 We can’t change the hospital source pressure but we can increase the delta between source and regulation pressure by going to larger-orifice PSOLs with lower working pressures. 

So, test data doesn't show a large difference in the performance between 50 and 100psi source pressure, so this might not be that big of a problem, plot below comparing 50psi source pressure and 100psi source pressure.  The regulated pressure is much more consistent but the flow trace doesn't seem that affected.
[image]

**Expected behaviour**
* Regulated pressure should stay close to the regulation setpoint.

**How do you know it is wrong**

See videos above, the pressure swings wildly during inhale with FiO2 set to 100%.

**Code pointers**

Not code, but the issue lies primarily in the small delta between the expected source pressure and the working pressure of the oxygen control system in the vent.

**Don't forget**
- [ ] Put the issue in a milestone
- [ ] Assign it to one of the projects
- [ ] Add appropriate labels
- [ ] Mention dependent/blocked issues if any

inceptionev, 2021-02-04:
[image]

Initial data from tonight's testing show that the 17psi configuration does not meet the 60lpm peak flow requirement.

However the software was not recalibrated for the PVQ31 PSOL, so further testing should be done.

inceptionev, 2021-02-04:
(not pictured) the IR2010 reg was also tried in this configuration and showed a very similar result.

inceptionev, 2021-02-05:
With the software properly recalibrated for the PVQ31-40 (4.0mm orifice PSOL) and the intermediate pressure set to 17psi (working pressure for the PSOL), we get comparable performance as compared to the UFlow PSOL at 45psi.  The peak flow is actually a little bit higher, though not by much.  It seems that if we find that the bouncy intermediate pressure is a problem, of we want to be more immune to supply pressure variations, this could be a viable solution.  Also this proves out the smaller IR1010 regulator performance, providing adequate flow even at the lower pressure setting, although for other reasons it we should will probably not use it (#952), but it does show that a regulator with similar flow characteristics should work.

[image]

esc-works, 2021-03-25:
Recommend sizing up the flow orifice on the PSOL. This sheet has some information how to calculate the required Cv value:
https://www.parker.com/parkerimages/schrader/cat/english/FRL4cvsg.pdf
Recommend making sure you're being consistent with Cv values for other potential suppliers that units are consistent.

martukas, 2022-06-20:
This ticket needs a better name.