RespiraWorks/Ventilator #996: Oxygen Subsystem: Tygon tubing not rated for pressure

inceptionev, 2021-01-28:
**Current behaviour**
* The 5894K34 1/4" ID Tygon tubing is rated for 5psi max and should not be used at the oxygen inlet pressure nor the intermediate regulated pressure. 
* It could be used at the patient inhale limb pressure but since these hoses would be easy to mix up, whatever solution used would probably be best applied to all the entire oxygen inlet pathway unless it would be impossible to mix up.

**Expected behaviour**
* Inlet path should be rated for the highest expected hospital supply pressure, at least 50psi
* Regulated pathway will probably be between 15 to 45psi depending on our design

**How do you know it is wrong**

* McMaster spec for the tygon tubing. https://www.mcmaster.com/5894K34/

**Code pointers**
Lives in this doc: https://github.com/RespiraWorks/Ventilator/tree/master/manufacturing/oxygen

Possible solution:
* use PU push-to-connect pneumatic fittings and tubing from the oxygen inlet to the regulator and from the regulator to the PSOL. It possible that the PU hose can be used from the PSOL to the inhale limb as well but I'm not sure if we would run into flow constriction problems at the lower pressure of the inlet limb. Any pneumatics people want to chime in here?

Rough calc using an online calculator shows (apologies for the mixed units):

- 0.2m of 6mm OD, 4mm ID PU tube at 50psi pressure, 60lpm flow (oxygen inlet to reg): 0.7% drop (should be OK)
- 0.2m of 6mm OD, 4mm ID PU tube at 15psi pressure, 60lpm flow (reg to PSOL, PSOL to inhale): 8% drop (probably not great)
- 0.2m of 8mm" OD, 5mm ID PU tube at 15psi pressure, 60lpm flow (reg to PSOL, PSOL to inhale): 2.5% drop (better)

Using 8mm OD PU tubing for all, the following part numbers are examples:

- 8mm OD 5mm ID PU tubing, opaque green: https://www.mcmaster.com/50315K25-50315K321/
- Inlet and IR2010 regulator 1/4" NPT to 8mm tubing push connect: https://www.mcmaster.com/5225K715/
- IR1010 regulator and PVQ31-xx-01N PSOL 1/8" NPT to 8mm tubing push connect: https://www.mcmaster.com/5225K714/
- IBV19M PSOL: 1/8" BSPP to 8mm tubing push connect: https://www.mcmaster.com/5225K507/
- SMC PVQ31-xx-01 1/8" BSPT to 8mm tubing push connect: https://www.mcmaster.com/5225K305/

Would love to have a person with pneumatics experience check over this.

**Don't forget**
- [ ] Put the issue in a milestone
- [ ] Assign it to one of the projects
- [ ] Add appropriate labels
- [ ] Mention dependent/blocked issues if any

inceptionev, 2021-01-28:
@tijanim @martukas I tagged this v0.3 hardware because I believe this makes breaks the v0.3 design for use with supplied oxygen, and because it might have an impact on the physical layout.  Feel free to move the tag if appropriate.

martukas, 2021-02-13:
@inceptionev can you confirm to have built a modification with those particular parts and it has fixed the problem?

inceptionev, 2021-02-16:
Can confirm this was built with the 8mm tubing and push-connect fittings in a v0.3 enclosed build and it solves the problem of the tubing popping off the fittings, and provides sufficient flow of >60 lpm.  Note that with this setup, the regulator is still not oxygen-safe (#952), so v0.3 with respect to oxygen is for for control demonstration only.

inceptionev, 2022-07-19:
@yusufcance @canooonn to select the tubing we need to connect the PSOL outlet to the oxygen venturi, we need to check the flow rate.  If we use the 8mm PU tubing, the internal diameter will be 5mm.  For an estimated 150mm of tubing length, what is the required pressure to achieve 120L/min flow rate (you may also need to consider the resistance of the venturi for this, or provide sufficient margin).  Most hospitals have around ~350kpa oxygen outlets, so assume we regulate this down to 200kpa, and we want to achieve > 120L/m at this pressure given the selected tubing and venturi.