RespiraWorks/Ventilator #1019: More custom tubing fittings

martukas, 2021-02-05:
**What has to be done**
There are multiple combinations of McMaster fittings that can be replaced by custom 3d printed parts. In particular:
* oxygen sensor +overpressure valve manifold needs one more twist on the end, towards the filter face
* exhale return has 2 elbows in sequence
* exhaust output path has 2 elbows in sequence

**How do you know it has to be done**
This **maybe** has to be done, but maybe not?

[image]

There is not enough vertical space under pneumatics panel for 2 elbow fittings to go end-to-end. This makes installing difficult.
It could be that in the next iteration we will have more space under the acrylic and these custom parts will not be necessary. 
However, it might be overkill to modify metal enclosure design to accommodate plastic. However, the space under the pneumatics tray might still be increased due to other more salient reasons, for example -- space for backup battery.
However, there are still advantages to doing custom fittings - fewer assembly components means fewer failure points; plastic is easy to get manufactured to spec and there is practically no advantage to using the McMaster fittings past the initial prototyping stage.

**Code pointers**
Should make sure you are working with the latest pneumatic assembly to get an idea for the sizing of these parts. Vertical distance should be easy to measure in the assembly.
The fittings can be shorter than the maximum constrained, just like the blower adapter, for example.

There can be a new directory and page under pneumatics for "custom tube fittings", which could have the whole lot of them in there, instead of making separate pages for each. 

**Don't forget**
- [ ] Put the issue in a milestone
- [ ] Assign it to one of the projects
- [ ] Add appropriate labels
- [ ] Mention dependent/blocked issues if any

tijanim, 2021-02-06:
Thinking simbple 3-d print u-fittings will work. We may need to expand the thru hole slightly to allow them to snake thru, but shouldn't be an issue

martukas, 2022-06-20:
This is awaiting a merge in #1038 . The branch may be taken over by a new contributor and merged. Though likely these components will change yet again before v0.4 is completed.