RespiraWorks/Ventilator #1020: Enclosed CAD for v0.3: front panel should have full PCB stack mounted to touchscreen

martukas, 2021-02-05:
**What has to be done**
Model full PCB(with Pi and Nucleo)+interim board+steppers stack attached to touchscreen.
Touchscreen needs to be remodeled to have explicit mounting points for Pi.

**How do you know it has to be done**
#789 shows how this is done. This is the preferred way of mounting the electronics in this design iteration, rather than trying
to drill mounting points into sheet metal.

**Code pointers**
* check with @inceptionev about standoff size
* there should be models of PCB in the repo
* there is a link to interim card repo from the electrical assembly page
* stepper driver board models can probably be found somewhere online?

**Don't forget**
- [ ] Put the issue in a milestone
- [ ] Assign it to one of the projects
- [ ] Add appropriate labels
- [ ] Mention dependent/blocked issues if any

inceptionev, 2021-02-08:
Link to PCB stack STEP: https://github.com/RespiraWorks/Ventilator/blob/master/pcb/rev1_export/20200424v2-RespiraWorks-Ventilator-Rev1.0-RC2-3D-STEP.step

this contains the PCB, the PI, and the Nucelo

inceptionev, 2021-02-08:
Individual components:

- Bottom Layer:
  - Touchscreen (no 3D model): [product page](https://www.amazon.com/gp/product/B07VNX4ZWY/ )
- PCB Layer:
  - PCB: [3D STEP](https://github.com/RespiraWorks/Ventilator/blob/master/pcb/rev1_export/20200424v2-RespiraWorks-Ventilator-Rev1.0-RC2-3D-STEP-NO-PI-NO-STM.step)
- PCB + 1 Layer:
  - Nucleo: [3D STEP](https://grabcad.com/library/nucleo-board-1) | [user guide with mechanical drawing](https://www.st.com/resource/en/user_manual/dm00105823-stm32-nucleo64-boards-mb1136-stmicroelectronics.pdf)
  - Raspberry Pi 4: [3D STEP](https://grabcad.com/library/raspberry-pi-4-model-b-1) | [mechanical drawing](https://www.raspberrypi.org/documentation/hardware/raspberrypi/mechanical/rpi_MECH_4b_4p0.pdf)
- PCB + 2 Layer:
  - Interim Board: [3D STEP](https://github.com/inceptionev/VentilatorRev1InterimDaughtercard/blob/master/VentilatorRev1InterimDaughtercard.step)
- PCB + 3 Layer: 
  - Stepper driver: [product page but no drawing or 3D model](https://www.st.com/en/ecosystems/x-nucleo-ihm03a1.html) | It has similar dimensions to the lower larger half of the Nucelo board, if that helps.
- PCB + 4 Layer: 
  - Same as PCB + 3 Layer

inceptionev, 2021-02-08:
Dimensions for Interim board: 
[image]

inceptionev, 2021-02-08:
- lower standoff: [Product page] (https://www.digikey.com/en/products/detail/w-rth-elektronik/971090151/6174614) | [Drawing](https://www.we-online.de/katalog/datasheet/971xx0xx1.pdf)

inceptionev, 2021-02-08:
- upper standoff: [Product Page](https://www.digikey.com/en/products/detail/raf-electronic-hardware/M2117-2545-AL/9836050) | [Drawing](http://ms22.product-config.net/cgs/output/81214673/raf_M2117-2545-al.pdf)

inceptionev, 2021-02-08:
Dimensions for PCB (yes there is a 0.1mm oopsie in the lower right mounting hole):
[image]

inceptionev, 2021-02-08:
Dimensions for the PCB (yes there is a 0.1mm oopsie in the lower right mounting hole, but luckily we don't use that for mounting):
[image]

KishanKoradiya, 2021-02-11:
Some other helpful links for reference:

* [Vendor product page for upper standoff](https://www.rafhdwe.com/catalog/male-female-standoffs?page=product&cid=male-female-standoffs-metric&id=m2117-2545-al)
* [Vendor product page with CAD for lower standoff](https://www.we-online.com/catalog/en/SSTIE_STEEL_SPACER_STUD_METRIC_INTERNAL_EXTERNAL)

PR #1027 (2021-02-15): Full PCB stack model
Things to note:
* Parts have been sourced from vendors as per the links given.
* Upper standoff 22mm did not have a freely available CAD file. So a slightly smaller one was used.
* Stepper drive expansion board is merely a stand-in. No models or dimensions are available.

Closes #1020
closed 2021-02-15 by martukas