RespiraWorks/Ventilator #1024: Testing

martukas, 2021-02-09:
**Current behaviour**
All tests `test.sh` script

**Expected behaviour**

**How do you know it is wrong**

It should be possible to build, test and deploy directly on the RPI hooked up to the nucleo.
This is likely how software updates will work for the final product and at this stage of prototyping it makes sens to be deploying like this as well.

**Code pointers**

```
==2138==ASan runtime does not come first in initial library list; you should either link runtime to your application or manually preload it with LD_PRELOAD.
```

**Don't forget**
- [ ] Put the issue in a milestone
- [ ] Assign it to one of the projects
- [ ] Add appropriate labels
- [ ] Mention dependent/blocked issues if any

asmodai27, 2022-06-02:
Appears fixed since deployment scripts update: #1188 . We need to confirm this.

martukas, 2022-07-09:
Yes, this works now.

closed 2022-07-09 by martukas