RespiraWorks/Ventilator #1026: Main power harness assembly instructions

martukas, 2021-02-10:
**Folder/Page:**
Need to fill out the section on the main power harness in [wiring assembly page](https://github.com/RespiraWorks/Ventilator/tree/master/manufacturing/electrical/wiring)

DIN female connector was identified in #966 and added to BOM already.

Looks like total harness length should be about 60mm.

Probably need some soldering instructions.

Probably should have some variety of shrink tubing.

**Timing considerations:**
Want to have this to close out the v0.3 mechanical documentation.

martukas, 2021-02-10:
Also need to identify the ideal part for this thing that grounds the enclosure. Used a car part here, but it should probably be a proper "ring" shaped thing. 

[image]

Hardware and lock washer will be in enclosure sub-assembly, so don't worry about that here.

martukas, 2021-02-10:
For main length, I used this 4mm shrink tubing:
https://www.amazon.com/uxcell-4mm-Shrink-Sleeve-Tubing/dp/B071VGQH97

For the smaller segments closer to the soldered part, it was probably one of these kits:
https://www.amazon.com/gp/product/B0722HN8SW

But it maybe that this could be done better than my hack job here....

inceptionev, 2021-03-04:
@martukas I think this ticket can close?

martukas, 2022-01-22:
I think this remains relevant for v0.4, where it may be somewhat different and also will benefit from more detailed instructions. Keeping it open but unassigning because it's now further downstream until we have enclosures.

martukas, 2022-06-26:
This has been partly addressed in #1259 
We can close this once we have physically confirmed correctness of assembly instructions and have added some pictures of the process.