RespiraWorks/Ventilator #1034: Include cooling fan in v0.3 build CAD and BOMs

martukas, 2021-02-15:
**Folder/Page:**
Need to add cooling fan to v0.3 CAD models and list it on the BOMs in readmes.

Should probably go in the enclosure page. Can list as optional, since there is no electrical connection for it in current PCB.

Old Alibaba links in spreadsheet bom are dead. Need to find a new source.

Specs were listed as "AXIAL FLOW FAN, 80x80x25, 12VDC, 37CFM, 2500RPM"

Also needs a fan guard.

Will eventually be superseded by #962

martukas, 2021-02-19:
Seems like these might work as models:
https://grabcad.com/library/80mm-fan--1
https://grabcad.com/library/80mm-fan-filter-1

martukas, 2021-02-19:
Potential parts for BOM:

Both of these are slimmer than the originally envisioned 25mm, which may be advantageous.
https://www.amazon.com/Evercool-80mm-20mm-2500-Ec8020m12ca/dp/B00JNMI6DE
https://www.amazon.com/Noiseblocker-NB-BlackSilentPro-PC-P-Ultra-Silent/dp/B0083A0BIA

And for the fan guard looks like what I was originally sent was this:
https://www.amazon.com/Computer-Filter-Grills-Aluminum-Stainelss/dp/B07LG41F1K

There appear to be different colors of this as well, which might match the case better. But I am not sure if this is the best one for as the exterior guard. Workmanship around the edges is a bit whack and the mesh is a bit flimsy. Might still be good for interior, to protect wires and tubing form getting caught in the fan.

Alternatives for fan guard:
https://www.amazon.com/Computer-Filter-Inches-Dustproof-Cooler/dp/B08FY8J9SC
https://www.amazon.com/Bgears-Cooling-Fan-Filter-80mm/dp/B00315C03G
https://www.amazon.com/AC-Infinity-Ventilation-Computer-Electronic/dp/B079PV274Q
https://www.amazon.com/OKGEAR-Steel-Filter-Grill-Black/dp/B00152QN08

The 3rd one actually looks pretty badass to me :)

martukas, 2021-02-19:
Digikey also has a bunch, and at good rates:
https://www.digikey.com/en/products/filter/fans-thermal-management/221?gclid=Cj0KCQiA4L2BBhCvARIsAO0SBdb_Ae0RdAnDt_vgW1AT1gDCZn6HNPJ-Qk3dYfLb-za3mb1QFzMFKnAaArfsEALw_wcB&s=N4IgjCBcoEwMwyqAxlAZgQwDYGcCmANCAPZQDaIMAHAOwAsADIgLpEAOALlCAMocBOASwB2AcxABfKUA&utm_adgroup=Finger%20Guards%2C%20Filters%2C%20Sleeves&utm_campaign=EN_Product_Fans%2C%20Thermal%20Management_MBR&utm_content=Finger%20Guards%2C%20Filters%2C%20Sleeves&utm_medium=cpc&utm_source=google&utm_term=%2Bfan%20%2Bguard

inceptionev, 2021-02-19:
#962 is a related issue, here are my comments from that ticket:

If we have space constraints, consider using a more powerful fan in the same size envelope, adding holes that allow airflow to cross the pneumatic tray, and providing an outlet for the airflow that opens from the space below the tray. This will encourage airflow across the stepper motors, which create a lot of heat. Also pay mind to the steady state temperature of the Raspberry Pi processor, which will throttle back on performance if it overheats.

Here is a fan that is the same size with 2x the flow and 5x the static pressure: https://www.digikey.com/en/products/detail/nmb-technologies-corporation/08025DA-12P-AA-00/9841655

inceptionev, 2021-02-19:
For the fan grills, it would be nice to have something with an actual dust filter to keep the inside from becoming all dusty.  That will mean needed a fan with a higher static pressure.  The first complete grills looks good.  The last one looks cool but when the holes are small, beware of stalling the flow.

martukas, 2021-02-19:
Ugh, second one is kinda ugly? :D @afarrington17 was suggesting we could do like modern gaming computer cases do and have one of those magnetic filters that could be taken off for cleaning.

I am guessing something like this?
https://www.amazon.com/287mm-Filter-Computer-Magnetic-2-Pack/dp/B07QBZGS7L

Though I guess this might also require a bit of accommodation in the enclosure. Making a 1-2mm recess may not be easy with the type of sheet metal we are currently using.

inceptionev, 2021-04-03:
Here's what I have selected:

- Fan: https://www.digikey.com/en/products/detail/nmb-technologies-corporation/08025DA-12P-AA-00/9841655
- Grate, filter media, and retainer: https://www.digikey.com/en/products/detail/orion-fans/GRM80-45/2621300

Zip file attached has STEP models for a fan and grate with the same size and footprint.

Note that the filter retainer, when assembled is 9.9mm tall.  So we should have the feet of the ventilator give at least a 20mm gap to the surface to allow for proper airflow.  If that is not desired we can mount the filter assembly on the internal side of the enclosure instead, which will reduce the required clearance but make it harder to replace the filter.

[image]

inceptionev, 2021-04-04:
Here's a cutout profile that you can use for the fan: [80mmFanCutout.zip](https://github.com/RespiraWorks/Ventilator/files/6254106/80mmFanCutout.zip)

(sorry it's a zip file because github doesn't accept .ipt files)

[image]

martukas, 2021-04-06:
This solution has been tentatively adopted in CAD. 
The changes have been checked-in in branch addressing #957 
The current solution is partly documented in #962 but does not preclude from using other fans/filter/covers guards.

Since the specific fans might still change subject to testing, the ticket remains open but is kicked down to later milestone.

martukas, 2022-11-20:
The fan is too thick and may interfere with components conencted to the edges of the PCB.
So we must consider a "slim" fan such as this one (10mm):
https://www.digikey.com/en/products/detail/cui-devices/CFM-8010-13-10/6194926

Or perhaps a 15mm one would be small enough and more powerful.