RespiraWorks/Ventilator #1040: Include power connector on v0.3 closure

martukas, 2021-02-17:
**What has to be done**
Enclosure page should include main power wire harness and relevant hardware for mounting it to closure panel.
Closure CAD should include a hole for a 4-pin DIN connector and relevant hardware.
There should be an assembly with the DIN connector and mounting hardware. This assembly should replace the plain closure part in the relevant super-assemblies.
There should be a picture of the physical assembly included.

**How do you know it has to be done**
There is currently no provision for power input in the general CAD.

**Code pointers**
DIN female is listed on the wiring page. Hardware should be either M2 or M2.5. With nuts and washers.

This is also related to #1026 but cad need not include wires. Connector is enough.

**Don't forget**
- [ ] Put the issue in a milestone
- [ ] Assign it to one of the projects
- [ ] Add appropriate labels
- [ ] Mention dependent/blocked issues if any

martukas, 2021-02-23:
The 4pin female DIN has been incorporated into closure panel. There may still be need for a barrel type connector, but we are missing a definitive (confirmed to fit and install) solution.

martukas, 2022-06-21:
Selection of connector was also discussed in #966

inceptionev, 2022-07-12:
Datasheet is here, for KPJX-4S: https://cloud.kycon.com/Catalog_PDF/KPJX-PM.pdf

martukas, 2022-07-27:
The power adapter plug is 12.10mm in diameter and can go about 5mm deep before requiring an even wider diameter. So this may have to go into thinner acrylic and the cutout size will have to be increased.

martukas, 2022-07-27:
[image]

closed 2023-03-21 by martukas