RespiraWorks/Ventilator #1041: Final drawing and exports update for v0.3

martukas, 2021-02-17:
**What has to be done**
Should create and/or update:
* DXF exports for all acrylic components, drawings and thumbnails
* STL exports for all 3d-printable components (drawings and thumbnails also preferred, on  best effort basis)
* Drawings and thumbnails for most other assemblies, on a best effort basis.
* General assembly could use a few renderings with some components selectively hidden to demonstrate the assembly process

Important considerations:
* make sure everything is in metric
* make sure all thumbnails are either 600 or 800 pixels in height, or whatever matches the renderings that they are next to in markdown tables
* most STLs should be up to date, and it may be confirmed by when the CAD vs. when the exports were last committed
* If assemblies are not complete or something makes it impractical to produce these, follow the priorities in the first list above to have these done by our milestone deadline.

**How do you know it has to be done**
Some of these are out of date or completely missing. In particular, there is nothing for the main pneumatics tray acrylic.
The others are there but might be out of date.

**Code pointers**
Crawl through all the pages in the manufacturing directory of the repo.

**Don't forget**
- [ ] Put the issue in a milestone
- [ ] Assign it to one of the projects
- [ ] Add appropriate labels
- [ ] Mention dependent/blocked issues if any

martukas, 2021-02-17:
The scope of this may be impacted by whether changes requested in the following tickets have been merged:
* sensor tubing #1039 
* main power input #1040 
* cooling fan #1034 
* custom elbows #1019 

As described in the main ticket body, this is on a best effort basis. The 4 components listed above are in order of priority.

tijanim, 2021-02-23:
### Drawings To be Created
***Parts in Bold are priority for drawing creation**
_*Italicized parts require .stl/.dxf exports_

- [ ] GA
     - [ ] Internals Assembly     
          - [ ] _**Pneumatics Tray Acrylic**_
          - [ ] **Equipment Bracket**
          - [ ] **Filter Panel Assembly**
               - [ ] **_Lower Faceplate_**
          - [ ] Pinch Valve Assembly
          - [ ] Blower Assembly
          - [ ] Venturi
          - [ ] **_Oxygen Manifold_**
          - [ ] Pinch Valve Assembly
          - [ ] **Upper Faceplate Assembly**
               - [ ] **_Upper Faceplate Acrylic_**
     - [ ] **Enclosure Assembly**
          - [ ] **Enclosure**

PR #1057 (2021-03-04): v0.3 Drawings and CAD Exports
### Drawings To be Created
***Parts in Bold are priority for drawing creation**
_*Italicized parts require .stl/.dxf exports_

- [ ] GA
     - [ ] Internals Assembly     
          - [x] _**Pneumatics Tray Acrylic**_
          - [x] **Equipment Bracket**
          - [x] **Filter Panel Assembly**
               - [x] **_Lower Faceplate_**
               - [ ] Filter Assembly
          - [ ] Pinch Valve Assembly
          - [x] Blower Assembly
               - [x] **_Support Acrylics_**
          - [ ] Venturi
          - [x] **_Oxygen Manifold_**
          - [x] ** [... cut]
closed 2021-03-04 by martukas