RespiraWorks/Ventilator #1061: Power loss alarm battery (AA) compartment selection

martukas, 2021-02-26:
**What has to be done**
There is an alarm for power loss that is autonomous and is powered by some AA batteries. These may have to be occasionally replaced. There should probably be an easily accessible compartment for these batteries on the back of the ventilator unit, and this compartment should be wired to the PCB.

We need to select this part so that we know how much space to provision for it in the enclosure and back acrylic panel. We need to do this prior to V0.4 enclosure design, so that we can:
* make sure it fits next to the PCB on the back panel
* make sure the opening in the sheet metal closure panel is big enough for this component among others

**Don't forget**
- [ ] Put the issue in a milestone
- [ ] Assign it to one of the projects
- [ ] Add appropriate labels
- [ ] Mention dependent/blocked issues if any

dcstraney, 2021-02-27:
Both the Keystone 2478 (http://www.keyelco.com/product-pdf.cfm?p=1040) and MPD BC4AAW (http://www.memoryprotectiondevices.com/datasheets/BC4AAW-datasheet.pdf) have a very common form factor and are available from multiple distributors.  Only thing to watch out for is the slightly different mounting hole spacings; Keystone's are countersunk and meant to use flat-head screws while MPD's are out of the direct path of the batteries and likely meant for some fairly low-profile such as pan heads.

martukas, 2021-03-25:
This has also been suggested:
https://complete-electronics.com/manufacturers/bulgin/battery-holders/bx0027bx0027

dcstraney, 2022-06-23:
Let's settle on the BX0027 for the foreseeable future, as the panel-mount "pull-out" form factor is really convenient?  (The other "classic" 4x AA form-factor ones can be the backup option, as those are easier to find swappable equivalents for, with a bit of thought to the mounting hole pattern)  It's currently available in reasonable-to-excellent quantities from a variety of distributors, so this isn't some hard to find, rarely-in-stock part.

martukas, 2022-11-10:
According to clearance holes in spec, mounting this would require m3
* measured mounting tab thickness = 2.4mm
* acrylic thickness = 6.35mm
* locknut = 4mm
* (optional 9mm OD washer = 0.8mm)

This totals 12.75-13.55mm. Therefore, mounting screws should be about 14mm.
Proposing this: https://www.mcmaster.com/92125A133/

martukas, 2022-11-10:
Locknut: https://www.mcmaster.com/93625A100/

closed 2023-03-21 by martukas