RespiraWorks/Ventilator #1077: Ventilator PCB Rev A Schematic Design Review

inceptionev, 2021-03-04:
This purpose of this ticket is to record and discuss the schematic design review of the Ventilator PCB Rev A.

We've split the pages up between engineers and this ticket can serve as a checklist.  If there is a finding that results in more than a minor or formatting change, please create a separate issue ticket for it and link it here.

- [ ] TOP (Stephanie)
- [ ] Logic/Control Power Supplies (Edwin)
- [ ] 5V UI/ / Control Supply (Edwin)
- [ ] 3.3V Logic / Sensor Supply (Edwin)
- [ ] Cycle Controller (Don)
- [ ] UI Computer (Edwin)
- [ ] Alarms and Power Monitoring (Stephanie)
- [ ] Indicators (Edwin)
- [ ] Safety FPGA (Edwin)
- [ ] Pressure Sensors (Stephanie)
- [ ] Pressure Sensor Daughterboard (Stephanie)
- [ ] Oxygen and Temperature Sensors (Don)
- [ ] NTC Temp Sensor (Don)
- [ ] Stepper Motor Driver (Edwin)
- [ ] Additional Aux Functions (Stephanie)
- [ ] System Monitor / FPGA ADC (Stephanie)
- [ ] Power Entry / Battery Handling (Don)
- [ ] Battery Charger / Source Switching (Don)
- [ ] Battery Balancer (Don)
- [ ] 12V System Buck Converter (Don)
- [ ] PSOL / Heater Driver (Edwin)
- [ ] STM Programmer for Cycle Controller (Stephanie)

We want to have a comprehensive design review before we go into layout.

Here are the original board requirements: https://docs.google.com/document/d/1RPkyGj1lCVv08fhpkGO5IUJaUcc130k_tNvB63ljrs0/edit?usp=sharing

Here's a schematic review checklist as a guideline, feel free to add to this:
- [ ] Transistors, diodes, and polarized caps oriented correctly.
- [ ] Pinout checks for all components
- [ ] Capacitor voltage ratings and voltage-capacitance derating
- [ ] Thermal margin for power components (see ticket 
- [ ] I2C nets not swapped and with one set of pullups, somewhere (remember that some modules like the RPi have onboard ones)
- [ ] MISO/MOSI directions and daisy-chaining, if applicable.
- [ ] JTAG direction and daisy-chaining, if applicable
- [ ] Single pin nets (apply NC / NoDRC mark if applicable)
- [ ] Input-output directions on ports and components
- [ ] Check for misspelled or incorrect power rail names
- [ ] Title block correctly filled out

**Don't forget**
- [ ] Put the issue in a milestone
- [ ] Assign it to one of the projects
- [ ] Add appropriate labels
- [ ] Mention dependent/blocked issues if any

inceptionev, 2021-03-05:
Potential changes in **bold**, everything else is just comments

**UI Computer:**
- Pinout check is good.
- **Doesn't look like +5V_RPi connects to anything (upper left of TOP sheet).  Should this be +5V_GUI?**
- **We've had trouble with powering the touchscreen exclusively from the pi's USB ports in the past, we should add a 2-pin micro-fit connector to provide 5V to the touchscreen independent of the Pi**
- **Let's flip the bit on UI_ID0 from 0 to 1.  The ID is meant to tell the software which version PCB it is plugged into, and 0b00 was rev 1.0.  Same goes for the Cycle Controller.**
- **Do we have a receptacle to receive the USB UART adapter TX/RX? (or are we even using it anymore? We wanted to have the Rpi be able to program the Cycle Controller, but can that be done through the STLink?)**
- **I think we might want the buzzer on one of the PWM pins**
- **The Rpi 3B+ never had well-published alternate pin functions, so we used generic GPIOs for CTS/RTS on PCB Rev 1,0, but the Rpi 4 does, [here](https://www.raspberrypi.org/documentation/hardware/raspberrypi/bcm2711/rpi_DATA_2711_1p0_preliminary.pdf), and it has dedicated UART0 CTS and RTS lines. CTS0 and RTS0 are on GPIO 16 and 17, respectively.**

inceptionev, 2021-03-05:
**Logic/Control Power Supplies**
- [x] General design looks good
- [ ] **CSD17304Q3 on 5V-USB-En: this should be a P-Ch FET for this work, also this is pretty oversized for the circuit, and we might be able to eliminate the level-shifter if we used a smaller, logic-level PFET (MTM231232LBF in SC70 package would do the trick.  Even better, I suspect we can lose this circuit altogether as we have been using the current Nucleo board with JP5 in the E5V position, and so this circuit was never exercised, and we had no troubles with programming.  Maybe leave the 5V-USB to diode-or into 5V-Ctrl so we have the option of powering the STLINK/CC/Sensors section of the board from USB when it is detached from the vent.**

**5V Buck Converter**
- [x] Pin check is good
- [ ] **Power Good is an open drain signal, so the testpoint won't have much meaning without a pullup**
- [ ] Inductor does not have a part selection so can't check power yet.
- [x] Diode: worst case with ambient at 40C, 70% duty cycle (16.8V VLink), diode Tjmax is 150C, and Rthja of 85C/W gives about 1.3W capability, which at Vf of 0.4V and duty cycle of 70% is 4.6A.
- [ ] **At Vlink of 16.8V, the 7.2uH inductor will give a ripple current of 1.2A, which feels high for our likely consumption of 2A on UI and <1A on CC.  10uH might be better.**
- [x] FB resistor divider yields 4.98V
- [x] Cap voltages on defined caps look good, but **100nF cap is not defined**

**3.3V LDO**
- [x] Pin check good
- [x] Cap voltages good
- [x] Power good in this case is a driven output, so is fine without a pullup

inceptionev, 2021-03-05:
**General**
- [ ] What, if anything, do we do about the battery charger needing a host to set charge current?  Won't charge the first time it is plugged in.
- [ ] Can we rename net oxygen level to oxygen_concentration or oxygen_fio2?  I feel like oxygen level makes it sound like a tank level.

inceptionev, 2021-03-05:
**Safety FPGA**

- [x] Pin check good
- [x] I2C polarity and pullups good

inceptionev, 2021-03-05:
**Indicators**

- [x] Simple page.  everything looks good here.  
- [ ] Not sure we need two power pins for 5V and GND, each of those microfit pins can carry 8+A.  Was the 10-pin connector selected to make it dissimilar from the other connectors (to avoid mistaken mating)?

dcstraney, 2021-03-05:
UI Computer:
- Touchscreen power: makes sense, done
- ID bits: same, for RPi.  **For the cycle controller I don't see any ID bits though, looks like it was never assigned in the ICD.  We may have to ditch the SD card to free up pins for this.**
- **Receptacle for USB UART: I don't know much about this - the STLink does have its own tx and rx but I couldn't find any info in the STLink manual about whether it exposes that as a generic USB serial interface.  Wouldn't hurt to have an extra connector for that.  Any preferred USB-serial cable? (I like the FTDI TTL-232R-3V3 personally but don't know if you guys already have a specific one on-hand)**
- Buzzer on PWM: yep makes sense, I noticed it wasn't when looking at the ICD but figured it was for a reason/holdover from last version. GPIO18/19 can also host PWM0/1 but with GPIO12/13 being freed up below...
- RTS/CTS: good find, changed.

Logic/Control Power Supplies:
- 5V-USB-En: Oops, I meant to copy over one of the P-FETs from the battery controller, looks like I grabbed the wrong one. (And yeah it's oversized, but didn't seem worth finding then sourcing an extra part for a one-off; same with the level-shifter even though it would be overcomplicating things in a different design; already had one for the blower)  Even better though if we can just ditch the power-switching bit though.  Yeah the point of the diode-OR is to be able to program the STM without powering up the whole thing, it was in our initial spec I think.
5V Buck:
- Power Good: yeah it's not actually intended for probing, it's just a "just-in-case" testpoint so in the rare event we actually need a power-good signal it won't be as hard as soldering to a DFN pin - figured we'd patch in a pull-up if we do happen to actually want to use it.
- Inductor: sounds reasonable, I just took that from the existing design, will use one in the same series as one of our existing ones.
- 100nF cap: not all the cap voltages are defined yet, that's still in process; I haven't picked out the USB mini-B connector yet either for the STM programmer.

General:
- Oxygen: sure, oxygen-FiO2 sounds reasonable (...-concentration seems overly long for typing out labels); changed.
- Pi power: good catch, fixed.  Deleted +5V_Rpi from library to prevent future mixups.

Indicators:
- Doubled pins: I agree we don't need 2x, will leave Stephanie to answer about the dissimilar pin #?

inceptionev, 2021-03-05:
**Actions from meeting:**
- [ ] Switch valve driver fuse to PTC
- [ ] Check in with software about how to do RPi to STM32 programming
- [ ] Switch to 100-pin version of STM32

inceptionev, 2021-03-06:
**PSOL Driver**

- [ ] **recommend enlarging lower FET bypass 0Ohm resistor to 0805 or bigger, to carry to up 2A.  Same with the high-side bypass 0Ohm resistor.**
- [ ] **recommend making the fuse a thermally resettable one, 0ZCF0200AF2C was selected for Rev 1.0**
- [x] cap voltage good
- [x] thermal: worst case Rds(on) @150C of 166mOhm, ambient temp of 40C, Rthja of 41C/W, yields max Pdiss of 2.7W, which equates to 4A of current.
- [x] pin check good
- [x] current limits and current sense resistor calculations good
- [ ] **Should we consider a resistor divider instead of the 10k current limiting resistor for the CS output?  Even weak overvoltages can saturate an ADC and affect the next reading in the conversion.**

dcstraney, 2021-03-07:
All sound reasonable; implemented.

dcstraney, 2021-03-07:
Changed the NTC temp sensor slightly to put the ESD protection on both pins just in case (might as well since one side was unused).  Can revert if there's a reason not to do this.

stephmusinsky, 2021-03-07:
swapping UI LED pinouts on ICD to match top level (just with each other)

stephmusinsky, 2021-03-07:
Pressure Sensor page: 
BMP280 symbol missing pin 7 to ground

stephmusinsky, 2021-03-07:
Additional Aux:
Found the I2C pull-ups I didn't see originally for CC-SDA/CC-SCL. I vote to leave on top page and take out here

stephmusinsky, 2021-03-07:
System Monitor ADC:
LM4120 symbol wrong: 5 should be labeled VOUT, but okay connections. Pin 1 named VRef (would be good to show it unconnected)

dcstraney, 2021-03-07:
All good catches, fixed

stephmusinsky, 2021-03-07:
STM Programmer:
Two pins have CC_SWCLK_OUT

dcstraney, 2021-03-07:
That's correct actually, it follows the nucleo STLink schematic (which is a massive pain to read because it's under-documented and has terrible net naming, but that's a separate story).

inceptionev, 2021-03-08:
**Stepper driver**

- [ ] pin check: everything correct, except symbol name is EPAD, but footprint name is 37.  Recommend fix this on the footprint.
- [x] IO wiring, LEDs, daisy-chain, all look good
- [x] Power wiring good
- [x] Cap voltages good

inceptionev, 2021-03-08:
Review things:

- [ ] Swap the indicators connector to 8 pin
- [ ] DNP the i2c pullup resistors on the FPGA page (the module has them already)
- [ ] check that the TVS diode on the NTC thermistor is correct
- [ ]

dcstraney, 2021-03-08:
From the first pass of thermal calcs it looks like the FETs can be made to work, but the DC (BATFET, Vin-Adp switches, Pack-P battery balancer switches) and buck high-side FETs (battery charger, +12V supply) definitely need some real heatsinking.  The buck low-side FETs are fine with the minimal footprint because duty cycles are always pretty high.  Any heatsinking scheme that gives a total Rth-ja of 70 K/W or less should work well for everything: can either...
1. Add lots of copper area, mechanical engineers probably won't be the biggest fan of this one but it might not be that bad space-wise
2. Via down and put a metal plate against the bottom of the board through some gap pad for electrical isolation
3. Use some SMT heatsinks like these guys: https://octopart.com/search?q=573100d00010g

inceptionev, 2021-03-08:
I'll take a look and see if we can find some lower Rds(on) options.  These aren't switching FETs so we don't care as much about Qg.  Might add BOM lines, but I think that is better than adding assembly steps.

inceptionev, 2021-03-09:
@dcstraney do you have your thermal calcs written down somewhere?  The only part I'm getting in my calcs as marginal is the P-Ch BATFET in the BQ24773 circuit, which we can definitely solve in the BOM by going to a 5x6mm part with lower thermal resistance and a bit less Rds(on).  Using the datasheet thermal resistance to ambient, with thermal vias to a backside ground plane, and a 35C ambient temperature, I get a temperature margin of 20C.  Not great, and we can get to a much better margin of 98C by going to SI7145DP-T1-GE3 (a 5x6 part).

All other parts have at least 30C of temperature margin mounted to the board with a solid backside ground plane and thermal vias as the only heatsinking.  The worst one is the BQ24773 because of its high switching frequency, and we are doing about as well as we can in the 5x6mm part class.  We do have the option to improve this by lowering the switching frequency from the default via i2c.

The DC FETs on the AC input have 44C of thermal margin.  I think that's good as I don't see our adapter current going up past the 120W level.

LM5141 switching FET has 53C of margin, and that's probably fine.  We could go up to a 100W blower and still be ok.  Lie you said, the bottom side FET is fine, because it's basically never on.

The FETs on the MAX17320 battery protector/balancer have 39C of margin.  This should be fine as-is, but we can open up the option for higher power blowers by going to a 5x6mm package here.

Calcs are here: https://docs.google.com/spreadsheets/d/1Otu4uRq1WR7CHx7AOSxF2eoAqoWKeeAT3UDNmPhiSdw/edit?usp=sharing

I'll get these changes (BQ24773 PFET and MAX17320 NFETs) into the schem shortly.

dcstraney, 2021-03-09:
Yeah, I took notes in a text file; it's the kind of thing that makes perfect sense to me but would be an embarrassing mess to anyone else so was going to check it against / put numbers into the spreadsheet later when I got a chance.
Anyways it looks like the main difference I can see is the different Rds-on values we're using - I'm using the max. values, plus accounting for the increase in Rds-on with temperature.  For the high-side buck CSD18563 for example the spreadsheet has 8.6 mOhms, or 10.8 mOhms max., but those datasheet values (for Vgs=4.5V) are at 25 C.  From figure 8 in the datasheet there's a ~1.6x multiplier @ 125 C, so if the final result does end up with Tj ~= 125 C then the real Rds-on is more like 17 mOhms.

inceptionev, 2021-04-21:
Review complete, design submitted.  Closing ticket.

PR #1101 (2021-04-26): Ventilator PCB Rev A
# Description

This is the design and layout of the Rev A PCB done on branch pcb_block_layout by @dcstraney, @stephmusinsky, and @inceptionev.  This design was reviewed on #1077 and layout completed on #1081 

Closes #1077 
Closes #1081 

# General checklist:

- [x] I have performed a self-review, including - looked through the `Files changed` tab, browsed repository in my branch
- [x] I have made corresponding changes to the documentation - to reflect changes in code, electrical or mechanical design
- [x] All new documentation or graphics follow the [documentation style g [... cut]
closed 2021-04-21 by inceptionev