RespiraWorks/Ventilator #1081: PCB Rev. A layout

dcstraney, 2021-03-12:
Placement and intra-section routing for each schematic section

Guidelines:
- For power and gnd connections, don't bother trying to route those on traces (except specific exceptions maybe in Edwin's battery sections) - it's safe to assume that pretty much everywhere you can and should just via down to hit either gnd or whatever power rail you want.  Power and gnd planes will be part of the global routing, after this is done.
- Remember that with multiple instances of the same schematic block, you can perform the whole layout on one instance, and copy it to the other instances using an "anchor" component for reference - make sure you have the plugins described here: https://github.com/RespiraWorks/Ventilator/wiki/Using-KiCAD#recommended-plugins
I (Don) am happy to walk you through the details of this.
- Testpoints are lumped into one block in the layout right now but should end up in whichever sections they belong to (except the ground testpoints, which should get placed at the end).

Sections which have non-obvious authorship/familiarity have a separate checkbox for whoever wants to claim them.  Feel free to edit this list and re-assign.
**Sections:**
Alarms / alarms.sch (Don)
* [ ] Placement
* [ ] Routing
Battery / battery_balancer.sch, battery_charger.sch, battery_controller.sch (Edwin)
* [ ] Placement
* [ ] Routing
Cycle Controller / cycle_controller.sch (Stephanie)
* [ ] Claimed
* [ ] Placement
* [ ] Routing
Solenoid Driver / driver_solenoid.sch (Don)
* [ ] Placement
* [ ] Routing
Stepper Driver / driver_stepper.sch
* [ ] Claimed
* [ ] Placement
* [ ] Routing
FPGA Daughterboard / fpga.sch (Stephanie)
* [ ] Placement
* [ ] Routing
Indicators / indicators.sch
(place @ end, as it's just one connector)
* [ ] Placement
Misc. I2C expander+EEPROM+Blower Control / misc.sch (Don)
* [ ] Placement
* [ ] Routing
Control Power / power.sch, power_3.3V_LDO.sch, power_5V_buck.sch (Don)
* [ ] Placement
* [ ] Routing
12V Blower Power / power_vsys.sch (Edwin)
* [ ] Placement
* [ ] Routing
O2 and Temperature Sensors / sensors_o2_temp.sch, sensors_temp_ntc.sch (Don)
* [ ] Placement
* [ ] Routing
Pressure Sensors / sensors_pressure.sch, sensors_pressure_daughterboard.sch (Don)
* [ ] Placement
* [ ] Routing
STM Programmer / stm_programmer.sch (Don)
* [ ] Placement
* [ ] Routing
System Monitor ADC for cycle controller / system_monitor_adc.sch (Edwin)
* [ ] Placement
* [ ] Routing
System Monitor ADC for FPGA / system_monitor_adc.sch (Stephanie?)
* [ ] Placement
* [ ] Routing
Raspberry Pi Connections / ui_computer.sch (Stephanie)
* [ ] Placement
* [ ] Routing
Ground clip testpoints
(place @ end)
* [ ] Placement

dcstraney, 2021-03-13:
Waiting to finalize placement of Alarms section until FPGA board outline is in place, as much of it will likely be sitting underneath FPGA board

stephmusinsky, 2021-03-20:
FPGA connector spacing:
on eval board- 2.5mm height and 6mm mating part
smallest matng connector- 11.5mm
So fpga eval board ~8mm above our pcb.

PMODs similar height 

usb height 4mm
power conn 7.5mm

stephmusinsky, 2021-03-20:
CR2016 heigh 1.6mm

stephmusinsky, 2021-03-20:
Battery holder 3.1mm height so CR2016 in it shouldnt be taller

inceptionev, 2021-03-29:
Don't mind me, just borrowing this thread to use as a checklist for remaining items:

- [ ] Place gnd clip testpoints
- [ ] Pour gnd all over top and bottom layers, inc. giving any existing top/bottom-layer planes a priority >0; via stitching for same
- [ ] Do the plane-cutout around the CS resistor sense connections, which I meant to do but forgot
- [ ] Minor change to the stepper drivers: use a local net in the schematic (like "+12V_Filt") instead of the "+12V_Step1" symbol for the filtered power, so that it doesn't expect a connection to be made between filtered power for one driver and filtered power for the other driver
- [ ] I'm also not a huge fan of the decoupling near the cycle controller, to be honest, but I don't think it's necessarily a dealbreaker.  Check out the STM programmer's decoupling cap placement
- [ ] Between increased current and looking @ the max Rds-on vs. typical, the high-side MOSFET for the battery charger is definitely in need of real heatsinking (everything else is fine) - the more copper area we can give it (and maybe avoid SMT heatsinks), the better.  How about adding some copper area - on the bottom, getting rid of the GND planes to the left and above the Buck-In plane, and the Net-(C109-Pad1) plane to its top-left, so that the Buck-In plane can expand?  On the top, can shrink the Net-(C109-Pad1) (see below) and move the high-side FET to expand the Buck-In plane
The low-side FETs even with minimal footprint (no drain copper area) still have reasonable margin, so might as well shrink the switch-node copper areas (Net-(C109-Pad1) and Net-(C146-Pad1), getting rid of their bottom-layer area too) as much as possible to get rid of an EMI hazard and free up area for the FETs that need it more

inceptionev, 2021-03-31:
Manufacturing package for the mainboard PCB has been submitted to to the fab/assembly vendor.  Consignment parts to be shipped shortly.

inceptionev, 2021-04-21:
Design package accepted by manufacturing.  Closing ticket.

PR #1101 (2021-04-26): Ventilator PCB Rev A
# Description

This is the design and layout of the Rev A PCB done on branch pcb_block_layout by @dcstraney, @stephmusinsky, and @inceptionev.  This design was reviewed on #1077 and layout completed on #1081 

Closes #1077 
Closes #1081 

# General checklist:

- [x] I have performed a self-review, including - looked through the `Files changed` tab, browsed repository in my branch
- [x] I have made corresponding changes to the documentation - to reflect changes in code, electrical or mechanical design
- [x] All new documentation or graphics follow the [documentation style g [... cut]
closed 2021-04-21 by inceptionev