RespiraWorks/Ventilator #1089: CI failure unrelated to recent code changes

dheater, 2021-03-22:
**Current behaviour/situation**

`controller-tests` are failing in CI, but I don't see any recent code changes that would have triggered the failures

**Expected state/behaviour**

CI tests passing

**How do you know it is wrong**

```
++ dirname software/controller/controller.sh
+ cd software/controller
++ uname -s
+ PLATFORM=Linux
+ '[' Linux '!=' Linux ']'
+ '[' --test == --help ']'
+ '[' --test == --install ']'
+ '[' --test == --clean ']'
+ '[' --test == --test ']'
+ pio test -e native
Verbose mode can be enabled via `-v, --verbose` option
Collected 16 items

Processing blower_fsm in native environment

Error: Unknown development platform 'native'

Exited with code exit status 1

CircleCI received exit code 1
```

**Starting points**

The most recent merge before the failures was #1080, but the tests were passing on the branch.

**Don't forget**
- [X] Put the issue in a milestone
- [X] Assign it to one of the projects
- [X] Add appropriate labels
- [X] Mention dependent/blocked issues if any

dheater, 2021-03-22:
The native controller-tests seem to be running fine on my local machine.

asmodai27, 2021-03-25:
Seems related to a bug in a new release of platformio : https://github.com/platformio/platformio-core/issues/3901
Will try to implement a fix ASAP. We may revert that change once platformio fixes their issue.

asmodai27, 2021-03-25:
Note that branches that had failing controller tests will need to be rebased once the above PR is successfully merged into master.
As of this writing, those are:
- https://github.com/RespiraWorks/Ventilator/tree/issue_957_equipment_tray_redesign (by @martukas)
- https://github.com/RespiraWorks/Ventilator/tree/pcb_block_layout (by @dcstraney)
- https://github.com/RespiraWorks/Ventilator/tree/803-migrate-from-valispace (by @dheater)

PR #1090 (2021-03-26): Explicitly install pio native platform for controller tests to run properly in CI
# Description

Explicitly install platformio "native" platform to get around a bug introduced in platformio 5.1.1 (see #1089 for a link to the pio issue)
Fixes #1089 : controller tests now pass on CI

# General checklist:

- [x] I have performed a self-review, including - looked through the `Files changed` tab, browsed repository in my branch
- [x] I have made corresponding changes to the documentation - to reflect changes in code, electrical or mechanical design
- [x] All new documentation or graphics follow the [... cut]
closed 2021-03-26 by asmodai27