RespiraWorks/Ventilator #1098: Air intakes on bottom of enclosure

martukas, 2021-04-20:
STUB-OUT! Fill this in later

**What is required**
Describe the desired feature, ideally in declarative form - what does success look like? This can be a user story.

**Subtasks:**
* [ ] Add subtasks or sections of content if applicable
* [ ] Identify documentation that should be updated upon completion

**How do you know it has to be done**
Link any documents, Slack threads, quote personal communication, reference your personal domain expertise etc. explaining why you think this should be done.

Link to requirements or risk analysis if applicable.

**Starting points**
Where might one start with solving this problem? Point to any relevant files or websites. Even if you have but a vague idea, please share. Perhaps something has prevented you or blocked you from solving the problem already?

**Don't forget**
- [ ] Put the issue in a milestone
- [ ] Assign it to one of the projects
- [ ] Add appropriate labels
- [ ] Mention dependent/blocked issues if any

inceptionev, 2021-04-20:
Notes from meeting:

- Intake should be filtered to prevent dust buildup
- Could use the same filter as the side fan openings, however these are 9mm tall, so they will either drive the need for tall feet on the bottom of the enclosure or they will have to be mounted on the inside of the enclosure.
- Filter can be inside or outside the enclosure, but outside is preferred for serviceability
- If a low profile filter is available, this would be good because it lessens the requirement for tall feet to provide a bottom gap beneath the ventilator.
- If possible, a snap in filter is good because it eliminates fasteners.

martukas, 2021-08-31:
Completed in 1117

PR #1117 (2021-08-31): Adding air intake grilles and technical drawings for enclosure
# Description

Added grille to the bottom. 
Added new technical drawings for enclosure parts.

Fixes #1098

# General checklist:

- [ ] I have performed a self-review, including - looked through the `Files changed` tab, browsed repository in my branch
- [ ] I have made corresponding changes to the documentation - to reflect changes in code, electrical or mechanical design
- [ ] All new documentation or graphics follow the [documentation style guide](https://github.com/RespiraWorks/Ventilator/wiki/Documentation-style-guide)
- [ ] Al [... cut]
closed 2021-08-31 by martukas