RespiraWorks/Ventilator #1118: Watchdog timer not connect on PCB

EnziinSystem, 2021-05-31:
I see watchdog timer STWD100NYWY3F not connect to anything IC while it must connect to Reset MCU pin.

Please view attach images
[image]
[image]

martukas, 2021-05-31:
@EnziinSystem which files are you referring to? Two versions of the PCB currently reside in the master branch of the repo, so a bit more precision in identifying would help.

inceptionev, 2021-06-15:
@EnziinSystem thanks so much for submitting an issue!  This is from a previous version of the PCB where we had an unconnected watchdog for experimentation.  The current version of the PCB (Rev A) uses the STM32 internal watchdog, so this circuit has been removed.

closed 2021-06-21 by inceptionev