RespiraWorks/Ventilator #1119: Not found RespiraWorks.lib and RespiraWorks_Std.lib

EnziinSystem, 2021-05-31:
When I open schematic in Kicad then I get the error message missing libraries.

RespiraWorks.lib and RespiraWorks_Std.lib

Please attach libraries to the project.

[image]
[image]

martukas, 2021-05-31:
Looks like our cikad libraries repo is held private for now, making it impossible for outsiders to work with the PCB design. This is a detractor from the "open-sourcedness" of our project.

EnziinSystem, 2021-06-03:
Hello,

Is there any way to solve this problem?

I think the component library is not a private license.

I just need to open the design diagram in KiCad.

Thanks.

inceptionev, 2021-06-15:
@EnziinSystem thanks so much for submitting an issue!  We are looking into this, and we should have it resolved soon.  We need to scrub our library to make sure we don't have resources with license conflicts with a public open-source license.  Once we have this resolved, we can make it public.

inceptionev, 2021-06-15:
The kicad project should have symbols and footprints embedded into it so it can be opened without access to the RespiraWorks library, it may ask you to "Rescue libraries" or something like this, and if you select OK, it will open.  I will try this on a fresh installation of kicad to confirm if it works.

inceptionev, 2021-06-28:
@EnziinSystem this library has now been made accessible.  https://github.com/RespiraWorks/KiCad-Library  Thanks for your ticket.

closed 2021-06-28 by inceptionev