RespiraWorks/Ventilator #1121: Pinch valve rotor press-fit has play

martukas, 2021-06-17:
**Current behaviour/situation**
Pinch valve rotor slides on too easily onto the stepper D-cut shaft and wobbles. Behaviour can be reproduced when printing with a relatively dimensionally stable resin such as Siraya Build or Siraya Fast. In this particular case, the rotors were printed vertical, but likely would still be an issue printed in other orientations.

The reason this problem may not have been seen before is that when printing the rotor vertical without supports, the bottom surface that is on the build plate will experrience some bloom due to increased exposure time on the first few layers. This will create a tighter fit when first inserting. However, the rest of the shaft interface is nevertheless too wide/loose. If exposure compensation is used for the bottom layers, or rotor is printed with scaffolding rather than directly on the build plate, or rotor is printed in another orientation, the problem will become apartent.

**Expected state/behaviour**
Rotor press-fit should be tight throughout the entire length.
Pinch valve Inventor models should be updated as well exported step files.

**How do you know it is wrong**

If the rotor is not well fit to the stepper shaft, it will wobble in operation and:
* calibration/linearization may not hold
* closed loop control behavior may become unstable
* rotor component may experience damage from repeated stress and eventually become even more loose, likely making the valve inoperable

**Starting points**
Should use manufacturer datasheet AND measure the shaft carefully with calipers. Should test print a few times to confirm that the dimensions ensure a tight fit.

**Don't forget**
- [ ] Put the issue in a milestone
- [ ] Assign it to one of the projects
- [ ] Add appropriate labels
- [ ] Mention dependent/blocked issues if any

martukas, 2022-01-22:
As of Summer 2021 this is still not solved. Might be 3d printer resolution limitations rather than design issues - i.e. would require a manufacturing method with better tolerances or perhaps reintroduction of set screw to stabilize the fit.

inceptionev, 2022-11-21:
This is probably a printing issue.  In the rotors we had printed for earlier units, it is tight all along the shaft and there is no wobble.  We also fitted the rotors on the shaft before post-curing, then cured the rotors while on the shafts.

I think we can solve this by adding these printing specifics to the documentation so that we get repeatable results.  For most other parts this is probably not needed, but for pressfit parts it may be necessary.