RespiraWorks/Ventilator #1125: Stainless steel to painted steel

gokcemay, 2021-07-13:
Please take the time to describe the issue in detail, so that other contributors can pick it up and implement or declare it done/obsolete in case it was fixed out of band.

**Current behaviour/situation**
* Enclosure and tray are made from stainless steel now. Stainless steel is 5 times more expansive than plain carbon steel, such as AISI 1020. We can change the design to AISI 1020. That will reduce cost, improve manufacturability and we can use color in design. 

**Expected state/behaviour**
* We have to check the standarts. Is there any reason not to use electrostatic paint on steel?

**How do you know it is wrong**

Stainless steel is expensive. Plain carbon steel is cheaper and used for manufacturing. It is easy to tool for and weights less. Also you can paint steel. That will solve design esthetics issue.

Link to requirements or risk analysis if applicable.

**Starting points**
You don't need to change any CAD files. We just change material and add paint info. The painted enclosure color is RAL 9016, internal tray is RAL 7035. Other RAL colors can be found here: [https://www.ralcolorchart.com/](https://www.ralcolorchart.com/)

[image]

**Don't forget**
- [x] Put the issue in a milestone
- [x] Assign it to one of the projects
- [x] Add appropriate labels
- [x] Mention dependent/blocked issues if any

martukas, 2021-07-20:
Also may be relevant - we have a branding guide with recommended colors based on RespiraWorks logo:
https://drive.google.com/file/d/14AHDmDAq0j-ZFKyvWVxcClCTFuS74Vht/view?usp=sharing

gokcemay, 2021-09-13:
Pricing doesn't change considirable amount for US based manufacturing.

closed 2021-09-13 by gokcemay