RespiraWorks/Ventilator #1126: PEM Nuts on acrylic is hard to assemble

gokcemay, 2021-07-13:
Please take the time to describe the issue in detail, so that other contributors can pick it up and implement or declare it done/obsolete in case it was fixed out of band.

**Current behaviour/situation**
* There are 6 PEM nuts on acrylic tray. These are fixed to internal tray. However tolerance buildup is an issue. Some bolts can't be assembeled.
* One way to fix it is to measure locations before cutting acrylic and fixing PEM nuts. 
* Another way to fix it is to increase hole sizes on internal tray. An increase to 5mm diameter might be enough.

**Expected state/behaviour**
* In order to change internal tray CAD files should be updated.

**How do you know it is wrong**

I tried to assemble it and found that there is a problem. Tighter production tolerances might solve the issue.

Link to requirements or risk analysis if applicable.

**Starting points**
Check tolerance buildup. What is the maximum issue we might have?

**Don't forget**
- [x] Put the issue in a milestone
- [x] Assign it to one of the projects
- [x] Add appropriate labels
- [x] Mention dependent/blocked issues if any

martukas, 2021-07-20:
Increasing hole sizes seems reasonable.

martukas, 2021-09-27:
Implemented in #1159

PR #1159 (2021-09-27): Equipment tray pem nut locations
# Description

This PR closes two issues. 
* One is because of tolerance build-up possibility 6 holes on the pneumatic tray might not align with the ones on equipment tray. Thus we need to increase hole sizes to ease alignment. That is done and holes on acrylic pneumatics tray are 1mm bigger in diameter. 
* The second issue was assembling and disassembling pneumatics tray was difficult. Bolts fit from bottom. Heat fitted nuts on acrylic were removed and PEM nuts are fitted to equipment tray to solve that issue. However M4x8mm bolts were short so 10mm in length  [... cut]
closed 2021-09-27 by martukas