RespiraWorks/Ventilator #1127: Nuts on acrylic is hard to assemble and dissasamble

gokcemay, 2021-07-13:
Please take the time to describe the issue in detail, so that other contributors can pick it up and implement or declare it done/obsolete in case it was fixed out of band.

**Current behaviour/situation**
* Acrylic panel has to be assembled and disassembled a few times to test the build. They are on acrylic, thus bolts have to be inserted from under the internal tray. There isn't a lot of space there.

**Expected state/behaviour**
* Acrylic and internal tray CAD files

**How do you know it is wrong**

Assembling the enclosure and tray showed it is hard to dis/assemble.

Link to requirements or risk analysis if applicable.

**Starting points**
A solution is to move nuts to internal tray. Another solution is to use the design as it is but put bolts from top. Thus use those bolts as indexing pins.

**Don't forget**
- [x] Put the issue in a milestone
- [x] Assign it to one of the projects
- [x] Add appropriate labels
- [x] Mention dependent/blocked issues if any

martukas, 2021-07-20:
Not totally clear on which ones are the problem. Can we have some pictures to illustrate?

martukas, 2021-09-27:
Implemented in #1159

PR #1159 (2021-09-27): Equipment tray pem nut locations
# Description

This PR closes two issues. 
* One is because of tolerance build-up possibility 6 holes on the pneumatic tray might not align with the ones on equipment tray. Thus we need to increase hole sizes to ease alignment. That is done and holes on acrylic pneumatics tray are 1mm bigger in diameter. 
* The second issue was assembling and disassembling pneumatics tray was difficult. Bolts fit from bottom. Heat fitted nuts on acrylic were removed and PEM nuts are fitted to equipment tray to solve that issue. However M4x8mm bolts were short so 10mm in length  [... cut]
closed 2021-09-27 by martukas