RespiraWorks/Ventilator #1140: Battery system behaves strangely with no battery attached

dcstraney, 2021-08-15:
PCB powered from 15V 1.67A wall wart for testing, no blower, no battery attached, no RPi or FPGA, bq24773 and MAX17320 are unprogrammed in their default states.  Pack-P voltage shows slow voltage ramp, with occasional oscillation from near 0V to near 15V.  See add'l discussion in "Sense Signals" section of PCB rev.A bring-up log: 
https://docs.google.com/document/d/1YLGYWPdSB4p-uytQrrjSSuHdnbYHyTh2L5PIqJrbATQ/edit#

[image]
[image]

inceptionev, 2021-08-16:
Reproduced condition on PCB SN 0.  Power supplies seem to come up normally, but BATT-P sweeps in a sawtooth, although on SN 0 it runs from about 6V to 9.3V with a frequency of ~3Hz.  This looks to be commanded action by the BQ24773, probably looking for the impedance of the battery.  You can see the brief turn-on command at the gate of the BAT PFET.

- Supply: Benchtop lab supply set to 15V, 1.7A
- Input power to pins 1 (GND) and 2 (15V) of J44
- Measured at pins 1-2: 15V, Vripple(p-p): 220mV
- Measured at pins 1-3: 13.6V, Vripple(p-p): 396mV
- Measured at 12Vout: 12.2V, Vripple(p-p): 133mV

inceptionev, 2021-08-16:
This doesn't necessarily seem to to be erroneous operation of the BQ24773 yet.  The BQ24773 is clearly switching the 15V input power down to 13.6V, which is then feeding the 12V buck converter.  The sawtooth is due to the action of the BAT PFET which is being commanded to turn on briefly every 300ms or so.  My guess is that if it senses no current, it switches the PFET back off.

The next step is probably to properly configure the IC for battery charging, and test it with a battery attached to see if the charging behavior is correct.