RespiraWorks/Ventilator #1144: Debug console user log

inceptionev, 2021-08-21:
Ticket to capture any weirdness that comes out using the new debug console.

Machine info:     uname_result(system='Windows', node='DESKTOP-B1DHPH0', release='10', version='10.0.19041', machine='AMD64', processor='Intel64 Family 6 Model 126 Stepping 5, GenuineIntel')

- [ ] Autocomplete doesn't work (windows 10, using mingw)
- [ ] connect list doesn't show any ports, although using the --port COMx switch does
- [ ] using controller_debug.sh --port COM5 doesn't forward the port switch

martukas, 2021-08-21:
How did you have to start it with specifying a python version? We might want to incorporate that into the bash script as well.

inceptionev, 2021-08-21:
To install the packages on windows I needed the following commands.  I have both python 2.7 and 3.8 and it kept installing into 2.7.  I tried pip3 first but that didn't seem to install into the right version.

`py -3.8 -m pip install matplotlib
py -3.8 -m pip install pandas
py -3.8 -m pip install GitPython`

inceptionev, 2021-08-22:
The command to run the program from windows console is:

`debug_cli.py --port COM5`

and in mingw:

`./debug_cli.py --port COM5`

both from the /utils/debug folder.

note that `/utils/controller_debug.sh --port COM5` runs but does not connect (perhaps because the flag is not passed?)

inceptionev, 2021-08-25:
request to expose forced_open and forced_off to command line interface

martukas, 2021-09-08:
Do `get` with wildcards, i.e. `get blower*` for all variables starting with `blower`