RespiraWorks/Ventilator #1194: ModuleNotFoundError: No module named 'git'

EnziinSystem, 2022-01-13:
When I run check in controller:

```
./controller.sh check
```

I get an error message:

```
Checking stm32 > cppcheck (platform: ststm32; board: custom_stm32)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Error: Processing stm32 (platform: ststm32; board: custom_stm32)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
ModuleNotFoundError: No module named 'git':
  File "/home/kevin/.platformio/penv/lib/python3.8/site-packages/platformio/builder/main.py", line 179:
    env.SConscript(item, exports="env")
  File "/home/kevin/.platformio/packages/tool-scons/scons-local-4.3.0/SCons/Script/SConscript.py", line 597:
    return _SConscript(self.fs, *files, **subst_kw)
  File "/home/kevin/.platformio/packages/tool-scons/scons-local-4.3.0/SCons/Script/SConscript.py", line 285:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "/home/kevin/Desktop/Working/Ventilator/software/controller/platformio/build_config/auto_firmware_version.py", line 24:
    import git
================================================================================= [FAILED] Took 0.32 seconds =================================================================================

Environment    Status    Duration
-------------  --------  ------------
stm32          FAILED    00:00:00.317
=========================================================================== 1 failed, 0 succeeded in 00:00:00.317 ===========================================================================
```

Thanks.

martukas, 2022-01-13:
It tells you exactly what the problem is - the git python module is not installed. The `controller.sh` script has an installation command that provides for all dependencies. Or you can probably install it manually with `pip`.

closed 2022-01-13 by martukas

EnziinSystem, 2022-01-13:
> It tells you exactly what the problem is - the git python module is not installed. The `controller.sh` script has an installation command that provides for all dependencies. Or you can probably install it manually with `pip`.

I installed and reboot but it still has an error.

```
./controller.sh install
sudo reboot
```
And test
```
./controller.sh test
Removing .pio
Processing stm32 (platform: ststm32; board: custom_stm32)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
ModuleNotFoundError: No module named 'git':
  File "/home/kevin/.platformio/penv/lib/python3.8/site-packages/platformio/builder/main.py", line 179:
    env.SConscript(item, exports="env")
  File "/home/kevin/.platformio/packages/tool-scons/scons-local-4.3.0/SCons/Script/SConscript.py", line 597:
    return _SConscript(self.fs, *files, **subst_kw)
  File "/home/kevin/.platformio/packages/tool-scons/scons-local-4.3.0/SCons/Script/SConscript.py", line 285:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "/home/kevin/Desktop/Working/Ventilator/software/controller/platformio/build_config/auto_firmware_version.py", line 24:
    import git
================================================================================= [FAILED] Took 0.36 seconds =================================================================================

Environment    Status    Duration
-------------  --------  ------------
stm32          FAILED    00:00:00.357
=========================================================================== 1 failed, 0 succeeded in 00:00:00.357 ===========================================================================
```

asmodai27, 2022-01-15:
Hi. If you are using the visual studio terminal, there might be a discrepancy between your systemwide python version and the one used by your IDE. Try running `which python` and `python - v` in both that terminal and a system terminal. If you get different answers, you will have to specify the correct pip for your IDE's version. In my case, I needed to run all pip commands from the installer with `pip3.8` in the IDE terminal at some point. This fortunately never happened again, even after a full réinstallation of my IDE.