RespiraWorks/Ventilator #1195: no operator "/" matches these operandsC/C++(349)

EnziinSystem, 2022-01-16:
I see messages warning/error in the test file:
```
no operator "/" matches these operandsC/C++(349)
```
It happens in all test files, please view the attached file.

[image]

asmodai27, 2022-01-16:
Hello

This message comes from the IDE not recognizing the includes in platformio.ini file.
When running the tests with platformio, the files are properly recognized, and (if it is the first time) platformio actually adds the proper includepath to the VScode environment, so this kind of error should disappear.

asmodai27, 2022-06-02:
Solution given above works on a fresh install of VSCode.

closed 2022-06-02 by asmodai27