RespiraWorks/Ventilator #1196: error: There is an unknown macro here somewhere

EnziinSystem, 2022-01-20:
Hello,
I run build gui on the **Rasberry Pi 4** with:

```
./gui.sh build
Removing build
Clean directory created: build
Checking app/main.cpp ...
Checking app/main.cpp: QT_DEBUG...
1/6 files checked 43% done
Checking src/gui_state_container.cpp ...
src/latching_alarm.h:65:3: error: There is an unknown macro here somewhere. Configuration is required. If Q_PROPERTY is a macro then please configure it. [unknownMacro]
  Q_PROPERTY(AlarmPriority::Enum nominalPriority READ GetNominalPriority CONSTANT)
  ^
Checking src/gui_state_container.cpp: QT_DEBUG...
```
Thanks.

martukas, 2022-01-22:
Looks like failure to have al the Qt libraries installed. This may also be affected by the version of RaspiOS that you are using. Since you are reaching out to us for quite a bit of fine-grained support lately, would you mind joining our Slack chat? Please email us at info@respira.works and we will gladly add you, we can then help you faster.

EnziinSystem, 2022-01-22:
@martukas 

Thank you for your support, I will contact you soon.

EnziinSystem, 2022-02-12:
@martukas 

I sent an email to info@respira.works, please join me to group Slack.

Thanks.

martukas, 2023-04-05:
Cannot reproduce this on our reference systems.

closed 2023-04-05 by martukas