RespiraWorks/Ventilator #1199: Refine venturi design using computational models

martukas, 2022-02-15:
**What has to be done**
Describe what has to be done, ideally in declarative form - what does success look like.

**Subtasks:**
* [ ] Add subtasks or sections of content if applicable
* [ ] Identify documentation that should be updated upon completion

**How do you know it has to be done**
Link any documents, Slack threads, quote personal communication, reference your personal domain expertise etc. explaining why you think this should be done.

Link to requirements or risk analysis if applicable.

**Starting points**
Where might one start with solving this problem? Point to any relevant files or websites. Even if you have but a vague idea, please share. Perhaps something has prevented you or blocked you from solving the problem already?

**Don't forget**
- [ ] Put the issue in a milestone
- [ ] Assign it to one of the projects
- [ ] Add appropriate labels
- [ ] Mention dependent/blocked issues if any

martukas, 2022-02-15:
Previous work on this question was done in #705

Other related tickets:
#631 
#1122 
#636

martukas, 2022-02-15:
Data files should be dumped on [Google drive](https://drive.google.com/drive/folders/1xd75hNHfbpOHuXmCEwh7YJWIuifTBUpx?usp=sharing)

yusufcance, 2022-03-12:
We did not want to change anything with the outer design because important design requirements were related with inside. To overcome with high speed and low pressure in venturi side, we operated part wider. The reason we did not make exact circle is to send more stable air for sensor to analyze.
	After creating new model, we put it to the parametric analysis to gather different result in different inlet conditions.
	Result on models are as fallows with less eddies, turbulence, and less difference between inlet and outlet by side of speed and pressure.

yusufcance, 2022-03-12:
[image]
[image]
[image]

yusufcance, 2022-03-12:
After creating new model, we put it to the parametric analysis to gather different result in different inlet conditions.
	Result on models are as fallows with less eddies, turbulence, and less difference between inlet and outlet by side of speed and pressure.

yusufcance, 2022-03-12:
[image]
[image]

yusufcance, 2022-03-12:
After all of these necessary informations and related datas we got from ANSYS, we would like to prepare plot which consists data of Pressure Difference on Sensors-mass Flow Rate on small region of tube. This analysis shows us we improved design by making a few changes inside of tube. This helps us to get close values im real-life tests.
[image]

inceptionev, 2022-06-22:
Results from the venturi comparison with the original design.  Data is at this link: https://docs.google.com/spreadsheets/d/1CylfLKrE64rkY6gbZZj0AzY_0tv3djxfM-mScOxUXCw/edit?usp=sharing

[image]

- The modified design produces less restriction of the flow, so the maximum flow is higher when the blower is turned up to its highest setting (9V for this setup)
- The sensitivity of the modified design is lower than the original, but neither get close to the full range of the 4kpa (40cmH2O) range of the dP sensor used in the ventilator design, and thus resolution is sacrificed, while also imparting a high pressure drop to the supply path of the ventilator
- This indicates to me that we should consider a lower-pressure dP sensor and a lower sensitivity venturi design to both reduce pressure drop and increase resolution.

inceptionev, 2022-06-22:
Here is a plot summarizing what performance goals we want for the venturi:

[image]

yusufcance, 2022-08-08:
Real-time value helped us to analyze our values in a more realistic way which can be easily compared with the computational value. As you can see from the graph upward, the value we got from the new design has approach the expected value when the inlet flow increases. But main target is to approach the curve as much as we can. It requires different modeling techniques to understand the outcome. Understanging of the flow here is the key factor because to analyze flow you have to understand the flow path, affects which are created by flow also the understanding of the venturi principle. To claim expected curve, we have used 5 different models and put them the ANSYS Fluent to see how flow get affected with different modeling techniques.

yusufcance, 2022-08-08:
1)	Slot Venturi Zone with Regular Outlet Pipe
 
[image]
[image]

yusufcance, 2022-08-08:
2)	Round-Small Venturi, Half-Regular Inlet and Regular Outlet
[image]
[image]

yusufcance, 2022-08-08:
3)	Smaller Venturi, Fully Regular Pipe
[image]
[image]

yusufcance, 2022-08-08:
4)	Smooth Transiton
[image]
[image]

yusufcance, 2022-08-08:
5)	Modified Smooth Transition
[image]
[image]

yusufcance, 2022-08-08:
Those are 5 different inner design and and flow analysis result. We gather all of the data to create parametric results. We had 2 real-time data, base and first modified design, which are not sufficient to create fully accurate parametrization. But with 5 new different data which are created in order will be very helpful and useful to create other options. Our main expectation is to get results from fifth design which may be more effective and accurate.
Theory of these design is to reduce rupture in the flow. By keeping more slope in transition we mostly provide that target. On the other side, we tried to keep clear sensor port inlets, that’s why there are difference in the first sensor pipe inlet between fourth anf fifth model. This ideology prevents air squeeze that helps us to get more accurate data. One other advantage is using smooth transition makes mathematical calculations easier. Because if you create more than one radius in the same curve, it will be harder on calculations. Analysing programs such as ANSYS create lesser need on mathematical calculations but to be certain it is also good to check mathematical side of the Venturi Principle.

inceptionev, 2022-08-16:
Measured data from 3D printed #5: Modified Smooth Transition:

[image]

inceptionev, 2022-08-16:
possibly useful paper on venturis: https://sci-hub.se/10.1016/j.petrol.2011.08.008