RespiraWorks/Ventilator #1208: Cleanup repo of binary stragglers

martukas, 2022-04-18:
**What has to be done**
There are still some PDFs, Power points and other random binaries in the repository. We need to move all of those out to appropriate areas in Google Drive.

**How do you know it has to be done**
Repo bloat and LFS bandwidth cost are still an issue. Prior to moving JPGs and PNGs back out into regular non-LFS space we should take care of any other binary files that may still be cluttering our repo.

**Don't forget**
- [ ] Put the issue in a milestone
- [ ] Assign it to one of the projects
- [ ] Add appropriate labels
- [ ] Mention dependent/blocked issues if any

martukas, 2022-07-07:
Repository history has been scrubbed using [git filter-repo](https://github.com/newren/git-filter-repo), in particular with:
```
git filter-repo --invert-paths --path-glob '*.pdf'
```

The following file types have been removed:
- *.zip
- *.step
- *.pdf
- *.PcbDoc
- *.SchDoc
- *.adicht
- *.stl
- *.SLDPRT
- *.xcf
- *.dat

...and some capitalization variations thereof.

The now eliminated binary types have also been untracked in LFS.

This is not to say that "dangling" commits will not have these files - i.e. commits that are not part of any live branches. This is so because our tickets and PRs may still reference these obsolete, i.e. unscrubbed commits. Consequently, they may still count towards our storage quota, but hopefully should not count to our bandwidth quota, because cloning and pulling should theoretically only bring in branches and commits referenced therein.

These assumptions appear to track, since a freshly cloned repo is now about 500MB with LFS, a bit less half that without it.

Further compactness may be hard to achieve other than by deleting the repository and starting a new one. This could still preserve the commit history for `master`, but would mean losing all tickets and PRs, so at this point may be too drastic.

More reading on this:
https://stackoverflow.com/questions/44161548/how-to-remove-dangling-commits-in-remote-git-repository-like-vsts
https://stackoverflow.com/questions/4367977/how-to-remove-a-dangling-commit-from-github
https://stackoverflow.com/questions/34579211/how-to-delete-a-file-tracked-by-git-lfs-and-release-the-storage-quota

We should file this under the "big lessons learned" about doing a hardware project with git. A project such as this ought to start with a very clear understanding of what types of files will need to be tracked and a clear strategy for management thereof. Then a configuration and procedure can be setup for what is allowed or forbidden. Something for the paper, @gokcemay @inceptionev ?

inceptionev, 2022-07-07:
PCBdoc and SCHdoc are Altium design source files.  Even though we have since gone over to KiCAD, these are the legitimate source design history of the v0.3 electronics.

martukas, 2022-07-07:
But, after v0.3 all work was in KiCAD, was it not? V0.3 is preserved.

inceptionev, 2022-07-07:
doesn't filter-repo remove their history from the repo?

martukas, 2022-07-07:
Altium files are back in. Still need to fix up deployment scripts to make sure graphics are pulled in.

martukas, 2022-08-11:
Ok, looks like the repo is clean of binaries now.

closed 2022-08-11 by martukas