RespiraWorks/Ventilator #1248: Finalize design of pack panel acrylic for v0.4

martukas, 2022-06-20:
**What is required**

The back-panel acrylic plate has a basic outline and a few basic features already modelled in Inventor. We should finalize it to include all of the following:

**Subtasks:**
* [ ] #1061 
* [ ] #961 
* [ ] #1040 
* [ ] USB pigtail from RasPI
* [ ] Ethernet pigtail from RasPI
* [ ] Oxgen inlet (already done)

Placement of all parts should observe where each of these connections will run to on the PCB or in the pneumatic system, to avoid wire harnesses being run in some awkward way or getting tangled.

**Starting points**
Existing Inventor model is a decent starting point.

martukas, 2022-06-22:
May also be impacted by #1254

zombiewater, 2022-07-27:
were going to add a part for the switch made of this thinner acrylic https://www.mcmaster.com/8560K191/

martukas, 2023-04-24:
#1318 completes this task, the 2 constituent sub-tasks that remain open can be worked-on and solved orthogonally.

closed 2023-04-24 by martukas