RespiraWorks/Ventilator #1257: Investigate fault-tolerance (FMEA)

dcstraney, 2022-06-24:
At some point we should do some serious systematic analysis and testing of how the design behaves in the case of various faults, and adjust the design as necessary to keep as much functionality as possible in single-failure cases - for example, we'd like the blower to keep running in as many situations as possible, and there should never be a "silent" failure that impairs functionality in some way without an alarm or notification.

It looks like there was a [higher-level FMEA](https://docs.google.com/spreadsheets/d/1qlTV5HqxnhlJXuhbSsEIfU-YwnN6PzgASv9_2dc5-cM/edit#gid=929514620) done before, which is plenty useful - but to know how good the design is at taking some hits we'll need to drill down further into details.

**Subtasks:**
* [ ] Read up on formal FMEA documentation in general, and any details specific to IEC 60601
* [ ] Identify and list potential failures to simulate
* [ ] Do design calculations to predict how system should behave in the case of each fault (for example: predicting DC input current if a power supply is stuck drawing its max. current limit).
* [ ] Test it out on a real prototype to check these predictions

**Starting points**
I'm not really qualified to comment on the mechanical parts, but the electrical failures to test include: over-current/shorts on each power rail, each power supply "failing open", individual external wires broken or entire external connectors unplugged, or input power brownout.  Software faults could include an infinite loop in a task.

Feel free to remove labels or split this into multiple issues if this is too much of a kitchen sink.

martukas, 2022-06-24:
This likely supersedes and encompasses #751 ?

dcstraney, 2022-06-24:
Yes good catch, closed out #751 for that reason but forgot to mention it here.